ZoneMinder/zoneminder #1369: Re-analyze an event

connortechnology, 2016-03-29:
I would like to be able to re-analyze an event.

For me right now, I want this because I used record mode for a few months and now wish that I had done mocord.

So I would like this implemented as a filter command, to re-analyze the images in the event and update the scores.

Ideally the code would also put in place the structure to analyze any dir or list of images.

knnniggett, 2016-03-29:
If we are going to do this, then I would propose we do what many manufactures call a "smart search".
Instead of analyzing the entire frame, the end user can select a zone that covers a portion of the image and the analyzing will only occur within that zone. 

This is very useful when the end user is searching for a specific incident within a field of view that happens to be very busy with motion. For example, searching a crowded room for the precise moment someone picks up an object.

Linwood-F, 2016-03-29:
One product I looked at had an interesting approach.  It saved all the analysis data from the frames (if you asked for this), and then it could do a search with different zones at very high speed as it did not have to deal with image data, only the analysis data.  It required saving more data of course, but it was very nice to then be able to zip through long periods of time very quickly.  I do not suggest this is possible in zoneminder's architecture -- I have no idea really -- just that if it is, it might bear consideration.  It is aimed more at "I do not know where my zone of interest is" than "I have no zones of interest -- oh, maybe I did" that you describe in the OP.   It's also conceivable if you want to re-analyze a month of frames, it is worth saving the data for the whole frame in case when you want to re-analyze, you are not quite sure where or what parameters.  Just a thought (not a volunteer, still haven't gotten back to my own work I wanted to do).

closed 2018-06-22 by stale[bot]