ZoneMinder/zoneminder #1370: Buffer key frames for h264 passthrough

SteveGilvarry, 2016-03-30:
Currently h264 starts recording at the same point as Zoneminder would grab a frame, however if this is not a key frame often videos will be blank until a keyframe occurs. Working on creating a buffer for avframes, at this stage trying STL queue.
Other issues that need attention based on this buffer
1. Video always starting on the previous keyframe and ZoneMinder recording frames based on alert.
2. Pre and Post event buffers

connortechnology, 2016-03-31:
Since video image rarely changes, I wonder if we could use the last known good keyframe?

I think though in general there isn't much we can do.  I feel like we should drop frames until we get a keyframe.  

In terms of pre and post event buffers, we just need to store frames.  I was thinking though that we could use a lot less memory if we stored the compressed frames instead of the uncompressed frames. Currently we are storing raw image frames.

SteveGilvarry, 2016-03-31:
Problem being as I understand it last key frame requires all the other
frames between it and alert frame to generate valid video aka uncompress
and display. Just for my problem on Record I need a way for event split to
occur on key frame as opposed to 600secs
In the case of h264 and jpeg disabled then no need for uncompressed frames,
instead we hold avpackets, which I believe are compressed frames. I was
originally thinking of them like network packets so many make up a single
frame, but docs say they contain compressed frame, be it I, B etc.

Rayn0r, 2017-04-16:
Today I started testing 1.30.2 feature-h264-videostorage and ran into the problem that the event video stream is a bit short. From a 15-20 second event only the last 5 seconds are captured. 
After some testing of increasing buffer size, with no real improvements, I started searching and stumbled upon this issue here.

My first question is: What is /dev/shm/zm.mmap.1 used for? It seems to be a ring buffer of some sort. After increasing "Image Buffer Size (frames)" to 500 frames (this should be 20sec at 25fps) this file jumped to almost 4GB... 
Why would it need to be so big? A 2 minute h.264 video is no bigger than 24MB at 16Mbit/sec (If I did the math right). So a 4GB buffer could store almost 3 hours of video... 
If the h.264 is actually written to this ring buffer, it should not be too hard to save the entire event to disk plus pre and post image count.

After figuring out the GOP size it should not be difficult to jump back a few I-frames and start saving the video from there to disk so that one could easily get the number of images set under "Pre Event Image Count".

Please correct me if I am wrong here.

connortechnology, 2017-04-17:
@Rayn0r please check out my storageareas branch.  I recently (last week) finally got around to finishing the code that implements a variable depth buffer.  At the moment I am finding that the stored video matches the event pretty much as well as can be managed.  

zm.mmap.1 is mostly a shared buffer of raw images.  I hope to transform my current h264 buffer into the shared buffer, so that zms can use the h264 packets for live viewing as well, but that will take some time.

It is so big, because it is raw images.  So 500 frames of 4 bytes times whatever the resolution is.  We will be transitioning it to storing compressed frames and maybe just a few raw frames, but that will take time.

Rayn0r, 2017-04-18:
> @Rayn0r please check out my storageareas branch. I recently (last week) finally got around to finishing the code that implements a variable depth buffer. At the moment I am finding that the stored video matches the event pretty much as well as can be managed.

Last night I checked out your storageareas branch and installed it. I must say that it works pretty well. 
I used Modect and the event is captured from the beginning till the end. It does seem however, that "Pre/Post Event Image Count" is not honored.

Thank you for the good work.

Rayn0r, 2017-04-18:
@connortechnology After switching to your storageareas branch I was always redirected to https://server/index.php?view=console upon login. This was kind of annoying because the path to my zoneminder installation is under https://server/zm/. Replacing the line below made it work correctly again:

```
diff --git a/web/index.php b/web/index.php
index d3a51de..1f049b6 100644
--- a/web/index.php
+++ b/web/index.php
@@ -213,7 +213,7 @@ if ( ZM_OPT_USE_AUTH && ! isset($user) ) {
 session_write_close();

 if ( $redirect ) {
-  header('Location: /index.php?view='.$view);
+  header('Location: '. dirname(parse_url($_SERVER['REQUEST_URI'], PHP_URL_PATH)) .'/index.php?view='.$view);
   return;
 }

```

I dunno if this is the most elegant approach, but it worked for me.

closed 2021-02-03 by connortechnology