ZoneMinder/zoneminder #1371: h264 Passthrough Remote Monitor

SteveGilvarry, 2016-03-30:
Enable h264 passthrough mode for Remote Monitors, should be pretty simple to extend the same process for ffmpeg to this code. Started work here, just haven't had time to debug why it is not recording.
https://github.com/SteveGilvarry/ZoneMinder/tree/video_remote

SteveGilvarry, 2016-07-27:
Note @connortechnology and I worked through this and we found that while remote is using the AVPacket struct it is only populating the data element, making passing this through to a container as a complete packet non-viable. So to enabled h264 passthrough on remote cameras needs more work on remote camera first.

connortechnology, 2021-02-03:
Just for the fun of it in 1.35.17 I made some more efforts to get this going.  In the end I had to disable it.  We have to populate the AVStream and AVCodecContext structs properly as well at least wrt time_base.  The remaining issue has to do with allocating/freeing the avpacket data. And I'm not sure that there is any reason to further maintain the remote RTSP code.  EXCEPT as an RTSP proxy server.

closed 2024-09-13 by SteveGilvarry