ZoneMinder/zoneminder #1372: h264 Passthrough Audio Recording

SteveGilvarry, 2016-03-30:
Some code in there to grab the audio packets, but I don't have a Mic on my camera and even then it is only doing G.711a and G.711mu. Streaming this content via HTML5 video is not supported by any browser so interested what codecs other peoples cameras support.
I am working on getting ffmpeg to stream a loop with audio so I can test this, but if @connortechnology has an audio capable camera and wants to have a go at it feel free.

I think the other option to consider attaching to this would be a flag to turn audio recording on and off, for places that forbid audio recordings. I know most cameras can control it but I am sure people will appreciate the option to trim audio packets out.

connortechnology, 2016-03-31:
So I did some work on this.  Pretty simple, add the flag and jsut don't store those packets if it isn't set. Still need to add it to the UI.  Will do that in a moment.

In my iconnor_video branch here I've done that and mostly just whitespace/documenting fixups.  

I do have cameras with audio, so I will test.

FrankieG, 2016-10-21:
Good day all, would it be outlandish to ask for audio to be recorded into a separate device all together.

Example:
Monitor one (front porch on first floor) obviously records the front porch as well as the audio associated with the monitor.

Monitor two (driveway) is located on the second story roof with a great view of the driveway and no audio.

The left side of Monitor one lines up with the right side of Monitor two.

By placing the audio recording as an "Audio Device" (mic or speaker) one could assign the audio channel to any monitor and/or multiple monitors allowing the audio picked up from monitor 1's mic of the conversation had between people standing in the driveway on monitor 2 and heard while viewing monitor 2.

I do realize there is a feature request location and apologize for placing this here however, I felt a good time to voice the option on here was better than late in the event a lot of code went into this already and made this very difficult to implement or not feasible.

knight-of-ni, 2018-09-23:
Since audio recording is supported in the latest 1.32.0 release, I am going to mark this as completed.

closed 2018-09-23 by knight-of-ni