ZoneMinder/zoneminder #1380: Make ZMS stream h264 directly from event video file.

connortechnology, 2016-04-01:


connortechnology, 2016-04-01:
Wwe can't be using a link directly to the video file.  Auth is required, and when storageareas gets put in, it just doesn't work.

So the link to video must be a link to zms.  So zms needs to be able to send the right stuff.

isusername, 2017-01-09:
https://gist.github.com/isusername/c627a73dab23fc90c6ac4eec6e85374e
I made this from these:
http://www.media-division.com/php-download-script-with-resume-option/
zoneminder/www/skins/classic/views/image-ffmpeg.php from the feature-h264-videostorage branch.

It can be used by replacing
`<?php echo getEventDefaultVideoPath($event) ?>`
with
`?view=video-event&eid=<?php` echo $eid;?>`
in event.php and timeline.php
It works fine with authentication turned on.

mastertheknife, 2017-03-23:
I think we can do that without zms.
A simple PHP script will do. We can use apache's mod_xsendfile to efficiently send the video file to the browser. Apache will take of everything, including partial transfers, making seeking possible.
I have done this at work but with MP3 files. it works well.

connortechnology, 2017-03-23:
It needs to authenticate.

connortechnology, 2017-03-23:
And we are moving away from apache long term.

mnoorenberghe, 2017-03-24:
Nginx has sendfile support too. I agree a simple PHP script doing auth. plus sendfile would be efficient and easy.

connortechnology, 2019-06-03:
I implemented this when implementing h265 support.

closed 2019-06-03 by connortechnology