ZoneMinder/zoneminder #1382: (FreeBSD) zms can process only 1 stream at the time

abishai, 2016-04-01:
I have an issue with zms when it streams
If I have 1 stream (for simplicity I paste into browser stream URL from camera http://10.0.4.6/cgi-bin/nph-zms?mode=jpeg&scale=100&maxfps=5&buffer=1000&monitor=3&connkey=806851&rand=1459541679 ) it plays well

If I copy that URL in another window, it waits until timeout.
If I stop the first tab, the second one immediately loads.
If I stop both tabs, none of them works on refresh throwning 502 error and zms		100236	FAT	Can't bind: Address already in use	zm_stream.cpp error.
If I delete zms-*.soc file, stream can be resumed with refresh button.

zms's sock file look strange - zms-806851s.soc (no k on the end)

knnniggett, 2016-04-01:
Working as intended.
This is not a bug, but a misunderstanding on how zms works. Each stream needs a unique connkey, but this is not the place to discuss how to manually build the url. Previous discussions can be found in the forums. You can of course ask us in IRC as well.

closed 2016-04-01 by knnniggett

abishai, 2016-04-02:
I'd like this issue to be reopened. The same applies to different connkeys and even cameras. So, I have 1 stream in montage, for example.

Also, stale soc files are left in SOCK dir. Is it normal?