ZoneMinder/zoneminder #1390: Bug: Events are not closed and remain as "new event" in the list

orwell96, 2016-04-04:
I set up ZoneMinder some days ago with one IP camera. Everything was working fine. Today I enabled email notifications, Since I did this, when new events are detected, a new event named 'new event' is created (with null/null images and 0/0 score, as always), but when the camera gets back into 'idle' state, the 'new event' remains and does not get updated with the actual event data.
After a few events it looks like this:
[image]
disabling email notifications and restarting ZM didn't change anything. Reverting recent changes I did before didn't change anything too. I think the email notifications are not related to this, the issue has to be in another place.

Maybe these log entries are related:
2016-04-04 21:33:33.996805	web_js		6601	ERR	getStreamCmdResponse stream error: Timed out waiting for msg /var/run/zm/zms-237430s.sock - checkStreamForErrors()	?view=watch	
2016-04-04 21:33:31.612777	web_php		6924	ERR	Timed out waiting for msg /var/run/zm/zms-237430s.sock	/usr/share/zoneminder/www/includes/functions.php	2371
2016-04-04 21:33:28.759338	zms		7095	ERR	Can't sendto on sd 9: No such file or directory	zm_monitor.cpp	3990
2016-04-04 21:33:28.610373	web_php		6693	ERR	socket_bind( /var/run/zm/zms-237430w.sock ) failed: Address already in use	/usr/share/zoneminder/www/includes/functions.php	2371

System: Debian 8.4
ZM v 1.29.0 (from Debian package repository)

orwell96, 2016-04-25:
I found the source of the issue:
Writing to the hard drive events should be stored to failed with an IO error. So Zoneminder couldn't save the event data. The event should be deleted and a warning be shown in this case.
The log entries are not related, they seem to appear every time you close an event playback when the stream hasn't started yet.

knnniggett, 2016-04-25:
Orphaned records are cleaned up by zmaudit. You can read about it in the documentation.
Additionally, an error is indeed logged if zoneminder cannot write to the either of the content dirs:
https://github.com/ZoneMinder/ZoneMinder/blob/master/web/index.php#L147

Naturally, one cannot guarantee orphaned records will be deleted or events logged if the IO error extends to the database.

closed 2016-04-25 by knnniggett