ZoneMinder/zoneminder #1391: debian/control at line 6: line with unknown format (not field-colon-value)

axklim, 2016-04-05:
`git clone https://github.com/ZoneMinder/ZoneMinder.git zoneminder`
`cd zoneminder/`
`sudo dpkg-checkbuilddeps 
dpkg-checkbuilddeps: error: syntax error in debian/control at line 6: line with unknown format (not field-colon-value)`

debian: 7.10

axklim, 2016-04-05:
`cat /etc/debian_version`
`gudik@sa-serv14:~/data/zoneminder/zoneminder$ cat debian/control 
Source: zoneminder
Section: net
Priority: optional
Maintainer: Isaac Connor <iconnor@connortechnology.com>
Build-Depends: debhelper (>= 9), autoconf, automake, quilt , libtool, dh-autoreconf
,libgnutls-dev | libgnutls28-dev
,libmysqlclient-dev | libmariadbclient-dev, libdbd-mysql-perl
,libjpeg8-dev|libjpeg9-dev|libjpeg62-turbo-dev
,libpcre3-dev
,libavcodec-dev, libavformat-dev (>= 3:0.svn20090204), libswscale-dev (>= 3:0.svn20090204), libavutil-dev, libavdevice-dev
,libv4l-dev (>= 0.8.3), libvlccore-dev, libvlc-dev
,libbz2-dev
,libsys-mmap-perl
,libnetpbm10-dev
,libcurl4-gnutls-dev | libcurl4-nss-dev | libcurl4-openssl-dev
,libgcrypt11-dev|libgcrypt20-dev, libpolkit-gobject-1-dev
Standards-Version: 3.9.4`

connortechnology, 2016-04-05:
I think I have fixed it.  When we do multi-line lines in the control file, the subsequent lines need spaces in front.

axklim, 2016-04-05:
it works! thx!

closed 2016-04-05 by knnniggett