ZoneMinder/zoneminder #1394: Make zones use a video stream instead of image

connortechnology, 2016-04-05:
Currently, when you load the zones page, zmu is called to grab an image and draw the zones over it.

Instead of doing that, how about we call zms and have it do it, providing a stream. 

Ideally, if the monitor is not recording, we could fall back to the most recent event.

kylejohnson, 2016-04-05:
I think that the still image in the Zones view makes sense.

connortechnology, 2016-04-06:
I have a camera that has something waving back and forth in front of it... it is hard to guage the range of motion of that movement in a still.

knnniggett, 2016-04-06:
What many applications do is, not only provide a live stream that @connortechnology has suggested, but many also have a visual indicator of the current motion as it compares to the current motion threshhold. 

So for example, you could have a bar graph that represents the current level of activity in realtime. The bar would also indicate the current threshold of motion. When the level goes above the motion threshold, the bar could changes colors to further emphasize that a motion event has occurred.

This would make tuning motion detection zones a lot easier.

connortechnology, 2016-04-06:
oh yeah... I hadn't gotten that far ahead in my thinking.  That would be so cool.

knnniggett, 2016-04-08:
Did some initial investigation on this. It was relatively easy to swap the zone image for a live stream.

While the individual points for each zone are drawn on the screen via php and overlaid on top of the image/stream, the hatch patterns are not. 

The hatch patterns are drawn and then embedded directly into a static jpg image here:
https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_monitor.cpp#L1114

What I am attempting to learn how to do is to change the behavior of DumpZoneImage to embed the zone information into an empty image, which has the same dimensions as the incoming stream, rather then embed the zone information into a snapshot of the incoming stream.

This, in turn, will allow us to overlay this new zoneimage over the top of the stream in the zone configuration page via php.

connortechnology, 2016-04-08:
I wonder if we could use svg to draw the zone over the image?  Cuz then it would be just a bit of code... very lightweight.

connortechnology, 2016-04-08:
apparently html5 has an svg tag... neato.

connortechnology, 2016-04-19:
closed by #1406

closed 2016-04-19 by connortechnology