ZoneMinder/zoneminder #1397: Zoneminder ONVIF (zmonvif-probe.pl) issues

lselinger, 2016-04-05:
After building/installing from source I mangaed to finally see the ONVIF options. After trying to run the probe from the GUI and not finding any cameras (I only have one ONVIF capable IP camera) I decided to try the probe manually (after using CPAN to grab all the required pms):
`# zmonvif-probe.pl probe -v
Use of uninitialized value in concatenation (.) or string at /usr/local/share/perl/5.18.2/SOAP/WSDL/Client.pm line 205.
Use of uninitialized value in concatenation (.) or string at /usr/local/share/perl/5.18.2/SOAP/WSDL/Client.pm line 205`

Checking SOAP::WSDL::Client.pm I see this:
`return $deserializer_of{ $ident }->generate_fault({`

Which is part of a larger if statement:

`    # if we had no success (Transport layer error status code)
    # or if transport layer failed
    if ( ! $transport->is_success() ) {

        # generate & return fault if we cannot serialize response
        # or have none...
        return $deserializer_of{ $ident }->generate_fault({
            code => 'soap:Server',
            role => 'urn:localhost',
            message => 'Error sending / receiving message: '
                . $transport->message()
        });
    }
} ## end sub call

1;`

So it looks like I'm not getting any response or nothing decent enough to get a good error. Looking for ideas to resolve.

siigna, 2016-06-16:
Hopefully this is the right spot.  I'm getting some syntax and "Global symbol requires explicit package name" errors.
ZM v1.30.0-RC1 on Debian wheezy with Perl v5.14.2 (Debian v5.14.2-21+deb7u3)

```
# ./zmonvif-probe.pl probe
syntax error at /usr/share/perl5/WSDiscovery10/Types/ProbeType.pm line 95, near "}{"
Global symbol "$attr" requires explicit package name at /usr/share/perl5/WSDiscovery10/Types/ProbeType.pm line 96.
Global symbol "$value" requires explicit package name at /usr/share/perl5/WSDiscovery10/Types/ProbeType.pm line 96.
Global symbol "$attr_str" requires explicit package name at /usr/share/perl5/WSDiscovery10/Types/ProbeType.pm line 99.
syntax error at /usr/share/perl5/WSDiscovery10/Types/ProbeType.pm line 102, near "}"
Compilation failed in require at (eval 90) line 2.
        ...propagated at /usr/share/perl/5.14/base.pm line 93.
BEGIN failed--compilation aborted at /usr/share/perl5/WSDiscovery10/Elements/Probe.pm line 18.
Compilation failed in require at (eval 89) line 2.
```

Any ideas?

blackPantherOS, 2016-06-17:
Same problem in 1.30.0:

### # zmonvif-probe.pl probe
```
Use of uninitialized value in concatenation (.) or string at /usr/lib/perl5/vendor_perl/5.22.1/SOAP/WSDL/Client.pm line 205.
Use of uninitialized value in concatenation (.) or string at /usr/lib/perl5/vendor_perl/5.22.1/SOAP/WSDL/Client.pm line 205.
```

### #zmonvif-probe.pl -v probe

```Probing for SOAP 1.1
Received message:
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:d3="http://www.onvif.org/ver10/network/wsdl/RemoteDiscoveryBinding" xmlns:d4="http://www.onvif.org/ver10/network/wsdl/DiscoveryLookupBinding" xmlns:dn="http://www.onvif.org/ver10/network/wsdl"><SOAP-ENV:Body><SOAP-ENV:Fault><faultcode>SOAP-ENV:VersionMismatch</faultcode><faultstring>Invalid SOAP message or SOAP version mismatch</faultstring></SOAP-ENV:Fault></SOAP-ENV:Body></SOAP-ENV:Envelope>
Error deserializing message. No message returned from deserializer.
Probing for SOAP 1.2
Received message:
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:d3="http://www.onvif.org/ver10/network/wsdl/RemoteDiscoveryBinding" xmlns:d4="http://www.onvif.org/ver10/network/wsdl/DiscoveryLookupBinding" xmlns:dn="http://www.onvif.org/ver10/network/wsdl"><SOAP-ENV:Header><wsa:MessageID>uuid:5e0b7340-c161-4f3e-9a15-248f06a1a597</wsa:MessageID><wsa:RelatesTo>urn:uuid:7AEFC286-348A-11E6-B2A8-B82B3739669A</wsa:RelatesTo><wsa:To SOAP-ENV:mustUnderstand="true">http://schemas.xmlsoap.org/ws/2004/08/addressing/role/anonymous</wsa:To><wsa:Action SOAP-ENV:mustUnderstand="true">http://schemas.xmlsoap.org/ws/2005/04/discovery/ProbeMatches</wsa:Action></SOAP-ENV:Header><SOAP-ENV:Body><d:ProbeMatches><d:ProbeMatch><wsa:EndpointReference><wsa:Address>urn:uuid:HZD-125019-AAEED-4adf-8702-3c330041408b</wsa:Address></wsa:EndpointReference><d:Types>dn:NetworkVideoTransmitter</d:Types><d:Scopes>onvif://www.onvif.org/type/video_encoder onvif://www.onvif.org/type/audio_encoder onvif://www.onvif.org/type/ptz onvif://www.onvif.org/location/global onvif://www.onvif.org/name/NetworkCamera onvif://www.onvif.org/hardware/HDIPC </d:Scopes><d:XAddrs>http://192.168.0.103:8000/onvif/device_service</d:XAddrs><d:MetadataVersion>1267264204</d:MetadataVersion></d:ProbeMatch></d:ProbeMatches></SOAP-ENV:Body></SOAP-ENV:Envelope>
l_xaddr = http://192.168.0.103:8000/onvif/device_service
http://192.168.0.103:8000/onvif/device_service, 1.2, (type='video_encoder', type='audio_encoder', type='ptz', location='global', name='NetworkCamera', hardware='HDIPC')```

SteveGilvarry, 2016-06-17:
1.30 Master on Ubuntu 16.04 seems to work no issues
./zmonvif-probe.pl probe
http://192.168.0.200/onvif/device_service, 1.1, (location/country='', name='Dahua', hardware='IPC-HFW3300C', Profile='Streaming', type='', extension='unique_identifier')

blackPantherOS, 2016-06-17:
The problem for me absolutly random, if run the command for more times I get good output, or sometimes half good 

monvif-probe.pl probe

Use of uninitialized value in concatenation (.) or string at /usr/lib/perl5/vendor_perl/5.22.1/SOAP/WSDL/Client.pm line 205.
http://192.168.0.103:8000/onvif/device_service, 1.2, (type='video_encoder', type='audio_encoder', type='ptz', location='global', name='NetworkCamera', hardware='HDIPC')
http://192.168.0.102:8000/onvif/device_service, 1.2, (type='video_encoder', type='audio_encoder', type='ptz', location='global', name='NetworkCamera', hardware='HDIPC')

knnniggett, 2016-06-17:
Just for the sake of knowing, those that are having issues, please report the version of perl's soap::wsdl module you are using:
```
[abauer@vfedora ~]$ rpm -qa |grep -i wsdl
perl-SOAP-WSDL-2.00.10-1.fc23.noarch
```

SteveGilvarry, 2016-06-17:
SOAP::WSDL	3.003

siigna, 2016-06-17:
libsoap-wsdl-perl, v2.00.10-1 on the wheezy system, v2.00.10-2 on the jessie system.

knnniggett, 2016-06-17:
I'm seeing a definite improvement after upgrading from soap::wsdl 2.00.10 to the latest 3.00.3

@jburgess777 I will have a soap::wsdl 3.003 rpm in the zmrepo testing by tonight. This fixes the issue you reported previously: https://github.com/ZoneMinder/ZoneMinder/pull/1411#issuecomment-208006474

@siigna you should try to find a 3.003 package. If this does not appear obvious, some of our Debian users, such as @SteveGilvarry , might be able to help you find the right package to install on your system. Recommend you avoid using cpan.

With all that said, I now get the same error @blackPantherOS reported so we are not out of the woods yet.

siigna, 2016-06-17:
> @siigna you should try to find a 3.003 package.
There's a 3.003 package in [stretch](https://packages.debian.org/stretch/libsoap-wsdl-perl)/[sid](https://packages.debian.org/sid/libsoap-wsdl-perl).  I'll backport it to wheezy/jessie and try it out.

blackPantherOS, 2016-06-17:
my SOAP-WSDL version is 3.003...

knnniggett, 2016-06-17:
see #1539 for another fix, helpful to those who reported receiving a perl syntax error

siigna, 2016-06-17:
ZM v1.30.0-RC1 on Debian wheezy with Perl v5.14.2-21+deb7u3, patch from #1539, libsoap-wsdl-perl v3.003-2 from stretch backported to wheezy:

```
# ./zmonvif-probe.pl -v probe
Probing for SOAP 1.1
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/SOAP/WSDL/Client.pm line 205.
Probing for SOAP 1.2
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/SOAP/WSDL/Client.pm line 205.
```

Debian jessie with Perl v5.20.2-3+deb8u5, patch from #1539, libsoap-wsdl-perl v3.003-2 from stretch backported to wheezy (I just installed the package I made in wheezy on jessie).

```
# ./zmonvif-probe.pl -v probe
Probing for SOAP 1.1
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/SOAP/WSDL/Client.pm line 205.
Probing for SOAP 1.2
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/SOAP/WSDL/Client.pm line 205.
```

knnniggett, 2016-06-17:
$transport->message() is undef, which causes the error. The hard part is understanding why this is happening.
Here is a clue:
At my office, where there are no Onvif devices on the lan, I get the uninitialised value error. At home, where I have several onvif devices on my lan, I now get valid results without any error messages.

siigna, 2016-06-17:
Thanks for the info!

I just had to check.  I've got an Axis M1011-W that I've been playing with.  I assumed ONVIF support would be enabled by default, but turns out it isn't.

After enabling from _Setup > System Options > Advanced > Plain Config > WebService > Enable Web Services_ (got this from an [Axis YouTube video](https://www.youtube.com/watch?v=UinkTCmW_Rs))
I'm still seeing the _uninitialized value_ error.

Weirdly enough, in the video you can see them searching the ONVIF.org conformant products list and the M1011-W is listed, but the current list makes no mention of it.  :confused:

Sending a _GetSystemDateAndTime_ request to it (grabbed an example from [stackoverflow](http://stackoverflow.com/a/22401873)) works:

```
# curl -s axis-m1011w/onvif/device_service -d '<s:Envelope xmlns:s="http://www.w3.org/2003/05/soap-envelope"><s:Body xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema"><GetSystemDateAndTime xmlns="http://www.onvif.org/ver10/device/wsdl"/></s:Body></s:Envelope>' | xmllint --format -
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:c14n="http://www.w3.org/2001/10/xml-exc-c14n#" xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd" xmlns:ds="http://www.w3.org/2000/09/xmldsig#" xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd" xmlns:wsa5="http://www.w3.org/2005/08/addressing" xmlns:xmime5="http://www.w3.org/2005/05/xmlmime" xmlns:wsrf-bf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:wsrf-r="http://docs.oasis-open.org/wsrf/r-2" xmlns:tan1="http://www.onvif.org/ver10/analytics/wsdl/RuleEngineBinding" xmlns:tan="http://www.onvif.org/ver10/analytics/wsdl" xmlns:tan2="http://www.onvif.org/ver10/analytics/wsdl/AnalyticsEngineBinding" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:tev1="http://www.onvif.org/ver10/events/wsdl/NotificationProducerBinding" xmlns:tev2="http://www.onvif.org/ver10/events/wsdl/EventBinding" xmlns:tev3="http://www.onvif.org/ver10/events/wsdl/SubscriptionManagerBinding" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:tev4="http://www.onvif.org/ver10/events/wsdl/PullPointSubscriptionBinding" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:timg="http://www.onvif.org/ver10/imaging/wsdl" xmlns:tptz="http://www.onvif.org/ver10/ptz/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:ter="http://www.onvif.org/ver10/error" xmlns:tns1="http://www.onvif.org/ver10/topics" xmlns:tnsaxis="http://www.axis.com/2009/event/topics">
  <SOAP-ENV:Body>
    <tds:GetSystemDateAndTimeResponse>
      <tds:SystemDateAndTime>
        <tt:DateTimeType>NTP</tt:DateTimeType>
        <tt:DaylightSavings>false</tt:DaylightSavings>
        <tt:TimeZone>
          <tt:TZ>GMT0BST,M3.5.0/1,M10.5.0</tt:TZ>
        </tt:TimeZone>
        <tt:UTCDateTime>
          <tt:Time>
            <tt:Hour>9</tt:Hour>
            <tt:Minute>4</tt:Minute>
            <tt:Second>53</tt:Second>
          </tt:Time>
          <tt:Date>
            <tt:Year>2011</tt:Year>
            <tt:Month>11</tt:Month>
            <tt:Day>10</tt:Day>
          </tt:Date>
        </tt:UTCDateTime>
        <tt:LocalDateTime>
          <tt:Time>
            <tt:Hour>9</tt:Hour>
            <tt:Minute>4</tt:Minute>
            <tt:Second>53</tt:Second>
          </tt:Time>
          <tt:Date>
            <tt:Year>2011</tt:Year>
            <tt:Month>11</tt:Month>
            <tt:Day>10</tt:Day>
          </tt:Date>
        </tt:LocalDateTime>
      </tds:SystemDateAndTime>
    </tds:GetSystemDateAndTimeResponse>
  </SOAP-ENV:Body>
</SOAP-ENV:Envelope>
```

knnniggett, 2016-06-18:
see #1541

blackPantherOS, 2016-06-18:
So, the ONVIF detection still listed the two camera after I replace the TransportUDP.pm but I can't setup I get this error message in log again:

`2016-06-18 18:43:07.288171	web_php		9970	FAT	Unable to probe network cameras, status is '255'	/usr/share/zoneminder/www/skins/classic/views/onvifprobe.php`

**zmonvif-probe.pl probe**
```
http://192.168.0.11:8000/onvif/device_service, 1.2, (type='video_encoder', type='audio_encoder', type='ptz', location='global', name='NetworkCamera', hardware='HDIPC')
http://192.168.0.12:8000/onvif/device_service, 1.2, (type='video_encoder', type='audio_encoder', type='ptz', location='global', name='NetworkCamera', hardware='HDIPC')
```

**zmonvif-probe.pl -v probe**

`Probing for SOAP 1.1
Received message:
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:d3="http://www.onvif.org/ver10/network/wsdl/RemoteDiscoveryBinding" xmlns:d4="http://www.onvif.org/ver10/network/wsdl/DiscoveryLookupBinding" xmlns:dn="http://www.onvif.org/ver10/network/wsdl"><SOAP-ENV:Body><SOAP-ENV:Fault><faultcode>SOAP-ENV:VersionMismatch</faultcode><faultstring>Invalid SOAP message or SOAP version mismatch</faultstring></SOAP-ENV:Fault></SOAP-ENV:Body></SOAP-ENV:Envelope>
Error deserializing message. No message returned from deserializer.
Received message:
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:d3="http://www.onvif.org/ver10/network/wsdl/RemoteDiscoveryBinding" xmlns:d4="http://www.onvif.org/ver10/network/wsdl/DiscoveryLookupBinding" xmlns:dn="http://www.onvif.org/ver10/network/wsdl"><SOAP-ENV:Body><SOAP-ENV:Fault><faultcode>SOAP-ENV:VersionMismatch</faultcode><faultstring>Invalid SOAP message or SOAP version mismatch</faultstring></SOAP-ENV:Fault></SOAP-ENV:Body></SOAP-ENV:Envelope>
Error deserializing message. No message returned from deserializer.
Probing for SOAP 1.2
Received message:
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:d3="http://www.onvif.org/ver10/network/wsdl/RemoteDiscoveryBinding" xmlns:d4="http://www.onvif.org/ver10/network/wsdl/DiscoveryLookupBinding" xmlns:dn="http://www.onvif.org/ver10/network/wsdl"><SOAP-ENV:Header><wsa:MessageID>uuid:fe81f304-244c-44b8-9e83-18efeda8d517</wsa:MessageID><wsa:RelatesTo>urn:uuid:FE851E5A-3572-11E6-80EA-1F463739669A</wsa:RelatesTo><wsa:To SOAP-ENV:mustUnderstand="true">http://schemas.xmlsoap.org/ws/2004/08/addressing/role/anonymous</wsa:To><wsa:Action SOAP-ENV:mustUnderstand="true">http://schemas.xmlsoap.org/ws/2005/04/discovery/ProbeMatches</wsa:Action></SOAP-ENV:Header><SOAP-ENV:Body><d:ProbeMatches><d:ProbeMatch><wsa:EndpointReference><wsa:Address>urn:uuid:HZD-125019-AAEED-4adf-8702-3c330041408b</wsa:Address></wsa:EndpointReference><d:Types>dn:NetworkVideoTransmitter</d:Types><d:Scopes>onvif://www.onvif.org/type/video_encoder onvif://www.onvif.org/type/audio_encoder onvif://www.onvif.org/type/ptz onvif://www.onvif.org/location/global onvif://www.onvif.org/name/NetworkCamera onvif://www.onvif.org/hardware/HDIPC </d:Scopes><d:XAddrs>http://192.168.0.12:8000/onvif/device_service</d:XAddrs><d:MetadataVersion>1267264204</d:MetadataVersion></d:ProbeMatch></d:ProbeMatches></SOAP-ENV:Body></SOAP-ENV:Envelope>
l_xaddr = http://192.168.0.12:8000/onvif/device_service
http://192.168.0.12:8000/onvif/device_service, 1.2, (type='video_encoder', type='audio_encoder', type='ptz', location='global', name='NetworkCamera', hardware='HDIPC')
Received message:
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:d3="http://www.onvif.org/ver10/network/wsdl/RemoteDiscoveryBinding" xmlns:d4="http://www.onvif.org/ver10/network/wsdl/DiscoveryLookupBinding" xmlns:dn="http://www.onvif.org/ver10/network/wsdl"><SOAP-ENV:Header><wsa:MessageID>uuid:22827385-d561-4a78-b635-514498174939</wsa:MessageID><wsa:RelatesTo>urn:uuid:FE851E5A-3572-11E6-80EA-1F463739669A</wsa:RelatesTo><wsa:To SOAP-ENV:mustUnderstand="true">http://schemas.xmlsoap.org/ws/2004/08/addressing/role/anonymous</wsa:To><wsa:Action SOAP-ENV:mustUnderstand="true">http://schemas.xmlsoap.org/ws/2005/04/discovery/ProbeMatches</wsa:Action></SOAP-ENV:Header><SOAP-ENV:Body><d:ProbeMatches><d:ProbeMatch><wsa:EndpointReference><wsa:Address>urn:uui [... cut ...] l onvif://www.onvif.org/name/NetworkCamera onvif://www.onvif.org/hardware/HDIPC </d:Scopes><d:XAddrs>http://192.168.0.11:8000/onvif/device_service</d:XAddrs><d:MetadataVersion>1267264204</d:MetadataVersion></d:ProbeMatch></d:ProbeMatches></SOAP-ENV:Body></SOAP-ENV:Envelope>
l_xaddr = http://192.168.0.11:8000/onvif/device_service
http://192.168.0.11:8000/onvif/device_service, 1.2, (type='video_encoder', type='audio_encoder', type='ptz', location='global', name='NetworkCamera', hardware='HDIPC')
`

knnniggett, 2016-06-18:
@blackPantherOS I still need a little more information from you. When you attempt this from the web console, during which step does it fail? After it detects the cameras and you click Next? Did you enter the cameras login credentials correctly before clicking Next?

Try calling this from the command line to see if zoneminder is able to retrieve the camera profiles:
`sudo zmonvif-probe.pl profiles "http://192.168.0.12:8000/onvif/device_service" 1.2 admin 12345`
and for your other camera:
`sudo zmonvif-probe.pl profiles "http://192.168.0.11:8000/onvif/device_service" 1.2 admin 12345`

Natrually, replace the login credentials with what you are using.

blackPantherOS, 2016-06-18:
_" After it detects the cameras and you click Next?"_
yes!
_"Did you enter the cameras login credentials correctly before clicking Next?"_
of course :)

**zmonvif-probe.pl profiles "http://192.168.0.12:8000/onvif/device_service" 1.2 MYUSER MYPASSWD**
`<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"/>`
**zmonvif-probe.pl profiles "http://192.168.0.11:8000/onvif/device_service" 1.2 MYUSER2 MYPASWD2**
`<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"/>`

knnniggett, 2016-06-18:
Well, that explains why it isn't working.
Stick a "-v" so we can see the full response.

blackPantherOS, 2016-06-18:
No, I tried already, the output is absolutly same not more.

knnniggett, 2016-06-18:
That's all I've got for the moment. We should get @altaroca 's take on this. The camera returns a fault when we query it for its profiles. Perhaps the camera doesn't fully implement onvif. Knowing the make and model would help. I guess the real question is, what can and what should we do when the camera returns a fault?

Note that, if you've got a Windows machine available, you might want to try the free Onvif Device Manager. https://sourceforge.net/projects/onvifdm/

It would be interesting to know if that tool can retrieve the camera's profiles.

blackPantherOS, 2016-06-18:
Cam type: KKMoon 8xx
Hardware Version:	JH41E-V1.0
Firmware Version:	10.8.79.8900 build 2015.06.09 02:51

product ex. on ebay [here](http://www.ebay.co.uk/itm/KKMOON-Wireless-Wifi-720P-HD-H-264-P2P-1MP-IP-Network-IR-Security-Camera-AD-G3K8-/152050196982?hash=item2366e5e1f6:g:IGwAAOSw9r1V-QbW)
A website with unofficial informations: [here](http://pathogenrush.blogspot.hu/2016/01/kkmoon-805-ipcamera-webviewer.html)

Not have any windows but i will try onvifdm with wine or in VirtualBox

blackPantherOS, 2016-06-18:
Ok I've tested cameras with onvifdm and working fine.
Do you want any information from onvifdm?

altaroca, 2016-06-19:
If the script does not return anything useful with the -v parameter then we need a network trace. @blackPantherOS a trace of the communication after `zmonvif-probe.pl profiles ...` would be really helpful. You can use wireshark or sth similar.

blackPantherOS, 2016-06-19:
Can I sending wireshark dump in email, if this good for you.
The dump just filtered to port 8000 enought for you?
The camera ip response in wireshark when I clicked 'Next' button after ONVIF detection: 
400 Bad Request

[A screenshot available here](https://s31.postimg.org/y5ij15au3/Screenshot_20160619_132311.png)

knnniggett, 2016-06-19:
@blackPantherOS 
First, filter your wireshark session by setting the destination or source ip to be that of the camera.
Second, right click one of the packets in the transfer, then do a Follow -> TCP Stream. Note: this feature is only available in recent versions of Wireshark, but it is well worth upgrading.

You can now step through the communication session, step by step, much like stepping through a video, frame by frame. It should be much more intuitive at this point. What you want to find is the SOAP message zoneminder sent to the camera, followed by the next SOAP response with the fault message in it.

blackPantherOS, 2016-06-19:
Wireshark is 2.0.4 ...
I have a better idea. I made a access to cam for you. Please write a email to me, and I will send login informations for you. what do you think?

knnniggett, 2016-06-19:
If you want your camera to be detected properly, then the burden is on you to do the leg work and provide the data. This is free support after all.

blackPantherOS, 2016-06-19:
Thaks to your time. I would like help your work, not important for me, because I controlling the p2p cameras from a mobile device.  The "leg work" not/never problem for me, just as you seeming my english not soo good, rather very bad:) So, I'm not sure that I can provide complex events which exactly for you. And we now working on and testing too much other applications on next release of blackPanther. I try reporting important errors for many other developers.  But I found a 10 years ago installed windows this mystic that still running, but this VirtualBox based "hunting" is a big s..cks. If this problem not solved later I will back after we released the next version from blackPanther.

knnniggett, 2016-06-20:
Welcome to the bleeding edge of software development. Onvif discovery is a new feature, still under development, and no promises have been made towards how many cameras it will work against. The only way to find that out is to try it.

Your English is just fine, and I'm a bit jealous of those who speak a second language. They didn't  emphasize that here in the US when I was in school. They should have, and if only I had the time to learn another language. If there is something you say which I don't understand, I will ask you. Alternatively, Google translate is pretty good. The language barrier can be overcome.

The few that choose to contribute to this project, do it in their spare time. We aren't getting paid, and we have far more work to do than time to do it. Consequently, I'm not going to be shy about spreading the load out. This is a community based project. The only difference between you and me is that I've hung around long enough to be trusted with the privilege to click the merge button at the button of the screen.

You've already demonstrated your ability to understand github and build ZoneMinder based off a recent pull request. Many don't get past that, but you did. Installing the latest version of Wireshark and clicking a few buttons is certainly within your skillset. You simply need to take the time to learn it. Asking questions is perfectly acceptable, and at the end of the day you'll have learned how to use a very important network analysis tool. 

You can choose to ignore this if you want, and in time the problem may or may not get fixed. It really depends on how widespread this problem is. Right now, all we know is it only affects your two cameras.

Tejasree27, 2017-02-10:
@knnniggett I have a similar issue. Zone minder shows Unable to probe network cameras, status is '255' after GetProfiles. I used the command that you mentioned above. Here is the output.

tejasreev@YYZ1DTP016:~$ zmonvif-probe.pl profiles "http://10.10.80.23:8080/onvif/device_service" 1.2 admin admin
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"><faultcode>SOAP-ENV:Server</faultcode><faultstring>Error deserializing message: unknown field type in ONVIF::Media::Types::Profile::_Profile::XmlAttr. Valid fields are:
token, fixed
Structure given:
$VAR1 = &apos;ONVIF::Media::Types::Profile::_Profile::XmlAttr&apos;;
$VAR2 = {
          &apos;token&apos; =&gt; &apos;0&apos;,
          &apos;type&apos; =&gt; &apos;tt:Profile&apos;,
          &apos;fixed&apos; =&gt; &apos;true&apos;
        };
 at /usr/share/perl5/ONVIF/Deserializer/MessageParser.pm line 212.
 at line 2 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 82.
. 
Message was: 
&lt;?xml version=&quot;1.0&quot; encoding=&quot;UTF-8&quot;?&gt;
&lt;SOAP-ENV:Envelope xmlns:SOAP-ENV=&quot;http://www.w3.org/2003/05/soap-envelope&quot; xmlns:SOAP-ENC=&quot;http://www.w3.org/2003/05/soap-encoding&quot; xmlns:xsi=&quot;http://www.w3.org/2001/XMLSchema-instance&quot; xmlns:xsd=&quot;http://www.w3.org/2001/XMLSchema&quot; xmlns:c14n=&quot;http://www.w3.org/2001/10/xml-exc-c14n#&quot; xmlns:wsu=&quot;http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd&quot; xmlns:xenc=&quot;http://www.w3.org/2001/04/xmlenc#&quot; xmlns:wsc=&quot;http://schemas.xmlsoap.org/ws/2005/02/sc&quot; xmlns:ds=&quot;http://www.w3.org/2000/09/xmldsig#&quot; xmlns:wsse=&quot;http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd&quot; xmlns:wsa=&quot;http://schemas.xmlsoap.org/ws/2004/08/addressing&quot; xmlns:wsdd=&quot;http://schemas.xmlsoap.org/ws/2005/04/discovery&quot; xmlns:chan=&quot;http://schemas.microsoft.com/ws/2005/02/duplex&quot; xmlns:wsa5=&quot;http://www.w3.org/2005/08/addressing&quot; xmlns:xmime=&quot;http://tempuri.org/xmime.xsd&quot; xmlns:xop=&quot;http://www.w3.org/2004/08/xop/include&quot; xmlns:tt=&quot;http://www.onvif.org/ver10/schema&quot; xmlns:trt=&quot;http://www.onvif.org/ver10/media/wsdl&quot;&gt;&lt;SOAP-ENV:Header&gt;&lt;wsse:Security SOAP-ENV:mustUnderstand=&quot;true&quot;&gt;&lt;wsse:UsernameToken&gt;&lt;wsse:Username&gt;admin&lt;/wsse:Username&gt;&lt;wsse:Password Type=&quot;http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest&quot;&gt;0KrceD52zqYab0s+n3kJQsip2XY=&lt;/wsse:Password&gt;&lt;wsse:Nonce&gt;YSB2YWx1ZSBvZiA1Njk=&lt;/wsse:Nonce&gt;&lt;wsu:Created&gt;2017-02-10T19:21:00Z&lt;/wsu:Created&gt;&lt;/wsse:UsernameToken&gt;&lt;/wsse:Security&gt;&lt;/SOAP-ENV:Header&gt;&lt;SOAP-ENV:Body&gt;&lt;trt:GetProfilesResponse&gt;&lt;trt:Profiles xsi:type=&quot;tt:Profile&quot; fixed=&quot;true&quot; token=&quot;0&quot;&gt;&lt;tt:Name&gt;Profile0&lt;/tt:Name&gt;&lt;tt:VideoSourceConfiguration xsi:type=&quot;tt:VideoSourceConfiguration&quot; token=&quot;0&quot;&gt;&lt;tt:Name&gt;VideoSourceConfiguration0&lt;/tt:Name&gt;&lt;tt:UseCount&gt;2&lt;/tt:UseCount&gt;&lt;tt:SourceToken&gt;0&lt;/tt:SourceToken&gt;&lt;Bounds xmlns=&quot;http://www.onvif.org/ver10/schema&quot; height=&quot;1080&quot; width=&quot;1920&quot; y=&quot;0&quot; x=&quot;0&quot; xsi:type=&quot;tt:IntRectangle&quot;&gt;&lt;/Bounds&gt;&lt;/tt:VideoSourceConfiguration&gt;&lt;tt:VideoEncoderConfiguration xsi:type=&quot;tt:VideoEncoderConfiguration&quot; token=&quot;0&quot;&gt;&lt;tt:Name&gt;VideoEncoderConfiguration0&lt;/tt:Name&gt;&lt;tt:UseCount&gt;1&lt;/tt:UseCount&gt;&lt;tt:Encoding&gt;H264&lt;/tt:Encoding&gt;&lt;tt:Resolution xsi:type=&quot;tt:VideoResolution&quot;&gt;&lt;tt:Width&gt;1920&lt;/tt:Width&gt;&lt;tt:Height&gt;1080&lt;/tt:Height&gt;&lt;/tt:Resolution&gt;&lt;tt:Quality&gt;4&lt;/tt:Quality&gt;&lt;tt:RateControl xsi:type=&quot;tt:VideoRateControl&quot;&gt;&lt;tt:FrameRateLimit&gt;30&lt;/tt:FrameRateLimit&gt;&lt;tt:EncodingInterval&gt;1&lt;/tt:EncodingInterval&gt;&lt;tt:BitrateLimit&gt;1024&lt;/tt:BitrateLimit&gt;&lt;/tt:RateControl&gt;&lt;tt:H264 xsi:type=&quot;tt:H264Configuration&quot;&gt;&lt;tt:GovLength&gt;11&lt;/tt:GovLength&gt;&lt;tt:H264Profile&gt;Baseline&lt;/tt:H264Profile&gt;&lt;/tt:H264&gt;&lt;tt:Multicast xsi:type=&quot;tt:MulticastConfiguration&quot;&gt;&lt;tt:Address xsi:type=&quot;tt:IPAddress&quot;&gt;&lt;tt:Type&gt;IPv4&lt;/tt:Type&gt;&lt;tt:IPv4Address&gt;224.1.0.0&lt;/tt:IPv4Address&gt;&lt;/tt:Address&gt;&lt;tt:Port&gt;40000&lt;/tt:Port&gt;&lt;tt:TTL&gt;64&lt;/tt:TTL&gt;&lt;tt:AutoStart&gt;true&lt;/tt:AutoStart&gt;&lt;/tt:Multicast&gt;&lt;tt:SessionTimeout&gt;PT0H0M0.060S&lt;/tt:SessionTimeout&gt;&lt;/tt:VideoEncoderConfiguration&gt;&lt;/trt:Profiles&gt;&lt;trt:Profiles xsi:type=&quot;tt:Profile&quot; fixed=&quot;true&quot; token=&quot;1&quot;&gt;&lt;tt:Name&gt;Profile1&lt;/tt:Name&gt;&lt;tt:VideoSourceConfiguration xsi:type=&quot;tt:VideoSourceConfiguration&quot; token=&quot;0&quot;&gt;&lt;tt:Name&gt;VideoSourceConfiguration0&lt;/tt:Name&gt;&lt;tt:UseCount&gt;2&lt;/tt:UseCount&gt;&lt;tt:SourceToken&gt;0&lt;/tt:SourceToken&gt;&lt;Bounds xmlns=&quot;http://www.onvif.org/ver10/schema&quot; height=&quot;1080&quot; width=&quot;1920&quot; y=&quot;0&quot; x=&quot;0&quot; xsi:type=&quot;tt:IntRectangle&quot;&gt;&lt;/Bounds&gt;&lt;/tt:VideoSourceConfiguration&gt;&lt;tt:VideoEncoderConfiguration xsi:type=&quot;tt:VideoEncoderConfiguration&quot; token=&quot;1&quot;&gt [... cut ...] ddress&gt;224.1.0.0&lt;/tt:IPv4Address&gt;&lt;/tt:Address&gt;&lt;tt:Port&gt;40000&lt;/tt:Port&gt;&lt;tt:TTL&gt;64&lt;/tt:TTL&gt;&lt;tt:AutoStart&gt;true&lt;/tt:AutoStart&gt;&lt;/tt:Multicast&gt;&lt;tt:SessionTimeout&gt;PT0H0M0.060S&lt;/tt:SessionTimeout&gt;&lt;/tt:VideoEncoderConfiguration&gt;&lt;/trt:Profiles&gt;&lt;/trt:GetProfilesResponse&gt;&lt;/SOAP-ENV:Body&gt;&lt;/SOAP-ENV:Envelope&gt;
</faultstring><faultactor>urn:localhost</faultactor></Fault>

Do you see what is the issue here? Thanks!

knnniggett, 2017-02-10:
what version of zoneminder are you running?

Tejasree27, 2017-02-10:
v1.30 on Ubuntu 14

Tejasree27, 2017-02-10:
The camera under test works fine with ONVIF Device Manager.

connortechnology, 2017-02-10:
1.30 is known to not support many onvif devices.  1.30.2 when released works with a lot more.

knnniggett, 2017-02-10:
Have you verified the URI `http://10.10.80.23:8080/onvif/device_service` applies to your camera? Each camera is different.  To do that you need to first execute `zmoncig-probe.pl probe` .

In any case, the upcoming 1.30.2 release contains new fixes to the onvif probe that we believe should resolve many, if not all of the issues. It properly detects all the cameras we own, which isn't a lot, but is definitely a big improvement. 

I'm going to close this particular issue out since we've fixed the issues specifically mentioned in it.

Please upgrade to the latest version when it becomes available. Then, if you still experience this issue, please report it in a new issue.

closed 2017-02-10 by knnniggett

Tejasree27, 2017-02-10:
Yes, this is the URI. As you can also see from the response that i posted. The camera gets detected in the zone minder and then on selecting the camera, GetProfiles is requested and then it faults on receiving the response. 

Anyways, I will check with the next release whenever it is.

knnniggett, 2017-02-10:
> Yes, this is the URI. As you can also see from the response that i posted. The camera gets detected in the zone minder and then on selecting the camera, GetProfiles is requested and then it faults on receiving the response.

Ah, you're right. There is one other thing you can do while you wait for the 1.30.2 package to become available for Ubuntu 14.04. Try doing the probe with no credentials at all. If it works, then like me you are the proud owner of a camera (likely from China) with no security on the ONVIF functions. If it doesn't work, then we are back to waiting.

Tejasree27, 2017-02-10:
I tired this. As you said, this camera does not have credentials yet and no security :)
It gets deteted but for GetProfiles, the error is the same though.