ZoneMinder/zoneminder #1398: Installing on Debian 7

axklim, 2016-04-06:
<pre>
gudik@sa-pc128:/data/work/zoneminder/zoneminder$ sudo dpkg-buildpackage 
dpkg-buildpackage: source package zoneminder
dpkg-buildpackage: source version 1.28.0-wheezy
dpkg-buildpackage: source changed by Isaac Connor <iconnor@connortechnology.com>
dpkg-buildpackage: host architecture amd64
 dpkg-source --before-build zoneminder
 debian/rules clean
dh clean --with quilt,autoreconf
   dh_testdir
   dh_auto_clean
   dh_quilt_unpatch
No patch removed
   dh_autoreconf_clean
   debian/rules override_dh_clean
make[1]: Entering directory `/data/work/zoneminder/zoneminder'
# Add here commands to clean up after the build process.
[ ! -f Makefile ] || /usr/bin/make distclean
dh_clean
make[1]: Leaving directory `/data/work/zoneminder/zoneminder'
 dpkg-source -b zoneminder
dpkg-source: warning: no source format specified in debian/source/format, see dpkg-source(1)
dpkg-source: info: using source format `1.0'
dpkg-source: warning: source directory 'zoneminder' is not <sourcepackage>-<upstreamversion> 'zoneminder-1.28.0'
dpkg-source: info: building zoneminder in zoneminder_1.28.0-wheezy.tar.gz
dpkg-source: info: building zoneminder in zoneminder_1.28.0-wheezy.dsc
 debian/rules build
dh build --with quilt,autoreconf
   dh_testdir
   dh_quilt_patch
No patches in series
   dh_autoreconf
autoreconf: 'configure.ac' or 'configure.in' is required
dh_autoreconf: autoreconf -f -i returned exit code 1
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
</pre>

SteveGilvarry, 2016-04-06:
> dpkg-source: warning: source directory 'zoneminder' is not - 'zoneminder-1.28.0'

Master just deprecated autotools should have cleaned up distro folders of
autotools installs.

knnniggett, 2016-04-06:
Use the debian_cmake distro folder.

knnniggett, 2016-04-06:
See #1399 and try again.

closed 2016-04-10 by knnniggett