ZoneMinder/zoneminder #1402: Passthrough for non h264

connortechnology, 2016-04-06:
So another option in some cameras is an mpeg4 rtsp stream.  

Passthough for these streams actually works ok, but the problem is that the browser <video> tag cann't handle an mpeg4 file.  It needs h264.  

So in order to view the event, zms would have to convert it. I think this is ok. Most events don't get watched, so it makes sense to not re-encode it while recording, and just re-encode it when viewing.

knnniggett, 2016-04-06:
This is one reason why I believe many surveillance applications need a plugin to be installed on the client browser in order for their product to work. While I am not particularly thrilled at this idea, it does move the burden of transcoding the stream from the server to client. This wouldn't matter much for small systems, but the larger the system you have the more important it becomes to not burden the server.

mattdurant, 2016-09-01:
The options for doing a client-side plugin are running out though.  ActiveX, Java, Flash are all deprecated out of or not supported in various browsers at this point.  The only real option that exists now for universal support is the HTML5 video tag and transcoding the event as it's streamed...

I've seen compelling examples of Javascript decoders for h264 and other codecs (none specifically for mpeg4), so it's entirely possible that a client-side decoder for the stream could be created that doesn't rely on plugins of any kind.

sagitt, 2018-10-05:
i have hikvision cameras that support rtsp audio stream

can zoneminder use this in live streaming or recording?

Source tab:

Remote Protocol: RTSP
Remote Method: RTP/RTSP
Remote Host Name: <username>:<password>@<ip_address>
Remote Host Port: 554
Remote Host Path: /live/ #/live.sdp/ with microphone connected
Target colorspace: 32 bit color
Capture Width (pixels): 1920
Capture Height (pixels): 1080
Preserve Aspect Ratio: Unchecked
Orientation: Normal
Deinterlacing: Disabled
Use RTSP Response Media URL: Checked

ASerbinski, 2018-10-25:
I haven't tested anything at all to do with this, but FWIW, I have a bunch of cameras that are able to produce h265 streams. That's with a FIVE on the end.