ZoneMinder/zoneminder #1405: php7 compatibility

onlyjob, 2016-04-08:
On _master_ [php7cc](https://github.com/sstalle/php7cc) utility reported the following:

~~~~
File: web/skins/classic/views/zone.php
> Line 32: String containing number in hexadecimal notation
    '0x00ff00';

File: web/skins/classic/views/timeline.php
> Line 563: Possible array element creation during by-reference assignment
    $frameSlots[$i][$j] =& $currFrameSlots[$i];

File: web/includes/functions.php
> Line 1314: Function argument(s) returned by "func_get_arg" might have been modified
    func_get_arg($i);
> Line 1325: Function argument(s) returned by "func_get_arg" might have been modified
    func_get_arg($i);

File: web/includes/database.php
> Line 84: Removed function "mysql_error" called
    mysql_error();

File: web/api/app/Controller/Component/ScalerComponent.php
> Line 8: Function argument(s) returned by "func_get_arg" might have been modified
    func_get_arg($i);
> Line 20: Function argument(s) returned by "func_get_arg" might have been modified
    func_get_arg($i);

File: web/ajax/zone.php
> Line 20: String containing number in hexadecimal notation
    '0x00ff00';
~~~~

I run _php7cc_ as `php7cc -x web/api/lib .` to exclude CakePHP from checking.
Other submodule(s) might need checking...

connortechnology, 2016-04-09:
I don't understand why we are even using func_get_arg in those places.  Totally messed up.

PR's coming.

connortechnology, 2016-04-09:
Oh, and the zone stuff just got a major rework by @knnniggett and I, so we can likely ignore those.

onlyjob, 2016-04-10:
Thank you, @connortechnology. I wonder if it would be possible to do a point release to address PHP-7 (in-)compatibility issues?

I applied #1409 on top of 0.29.0 but still got the following warnings:

~~~~
File: ZoneMinder-1.29.0/web/ajax/zone.php
> Line 20: String containing number in hexadecimal notation
    '0x00ff00';

File: ZoneMinder-1.29.0/web/skins/classic/views/zone.php
> Line 32: String containing number in hexadecimal notation
    '0x00ff00';

File: ZoneMinder-1.29.0/web/skins/classic/views/timeline.php
> Line 563: Possible array element creation during by-reference assignment
    $frameSlots[$i][$j] =& $currFrameSlots[$i];

File: ZoneMinder-1.29.0/web/skins/xml/views/console.php
> Line 87: Removed function "split" called
    split(',', $group['MonitorIds']);

File: ZoneMinder-1.29.0/web/includes/functions.php
> Line 1313: Function argument(s) returned by "func_get_arg" might have been modified
    func_get_arg($i);
> Line 1325: Function argument(s) returned by "func_get_arg" might have been modified
    func_get_arg($i);

File: ZoneMinder-1.29.0/web/api/app/Controller/Component/ScalerComponent.php
> Line 8: Function argument(s) returned by "func_get_arg" might have been modified
    func_get_arg($i);
> Line 20: Function argument(s) returned by "func_get_arg" might have been modified
    func_get_arg($i);
~~~~

Good news is that _php7cc found no problems in `web/api/app/Plugin/Crud`.  :)

connortechnology, 2016-04-12:
Not sure what we can do about line 32, the hexadecimal thing. We are using it correctly, as a string. Good news is that in 1.30 it will go away entirely.
Likewise the xml skin is gone now. 
I'm not sure what to do about functions.php line 1313.  We are no longer altering arguments...

onlyjob, 2016-05-07:
Thank you, Isaac. I think we can consider it fixed. :)

closed 2016-05-07 by onlyjob