ZoneMinder/zoneminder #1413: Installing on Debian 7

axklim, 2016-04-11:
<pre>
make[2]: Leaving directory `/data/work/zoneminder/zm/obj-x86_64-linux-gnu'
install -D -m 0644 debian/apache.conf  debian/tmp/etc/zm/apache.conf
rm debian/tmp/usr/share/zoneminder/api/lib/Cake/LICENSE.txt
rm: cannot remove `debian/tmp/usr/share/zoneminder/api/lib/Cake/LICENSE.txt': No such file or directory
make[1]: *** [override_dh_auto_install] Error 1
make[1]: Leaving directory `/data/work/zoneminder/zm'
make: *** [binary] Error 2
</pre>

I added in distros/debian/rules :
override_dh_auto_install:
	dh_auto_install --buildsystem=cmake
	install -D -m 0644 debian/apache.conf  $(INSTDIR)/etc/zm/apache.conf
	rm **-f** $(INSTDIR)/usr/share/zoneminder/api/lib/Cake/LICENSE.txt
	rm **-f** $(INSTDIR)/usr/share/zoneminder/api/.gitignore
	rm -r**f** $(INSTDIR)/usr/share/zoneminder/api/lib/Cake/Test

And:
<pre>
gudik@sa-pc128:/data/work/zoneminder$ sudo dpkg -i zoneminder_1.28.108-nmu2015100101_amd64.deb Selecting previously unselected package zoneminder.
(Reading database ... 195358 files and directories currently installed.)
Unpacking zoneminder (from zoneminder_1.28.108-nmu2015100101_amd64.deb) ...
dpkg: dependency problems prevent configuration of zoneminder:
 zoneminder depends on libsys-meminfo-perl; however:
  Package libsys-meminfo-perl is not installed.
</pre>

I do not have package named: libsys-meminfo-perl
<pre>
gudik@sa-pc128:/data/work/zoneminder$ cat /etc/debian_version 
7.10
gudik@sa-pc128:/data/work/zoneminder$ aptitude search libsys | grep perl
i   libsys-cpu-perl                 - module for getting CPU information        
p   libsys-cpu-perl:i386            - module for getting CPU information        
p   libsys-cpuload-perl             - module which retrieves load average of a m
p   libsys-cpuload-perl:i386        - module which retrieves load average of a m
p   libsys-filesystem-perl          - Perl module to retrieve list of filesystem
p   libsys-gamin-perl               - Perl interface to Gamin (File Access Monit
p   libsys-gamin-perl:i386          - Perl interface to Gamin (File Access Monit
i A libsys-hostname-long-perl       - Figure out the long (fully-qualified) host
i   libsys-mmap-perl                - module for using POSIX mmap               
p   libsys-mmap-perl:i386           - module for using POSIX mmap               
p   libsys-sigaction-perl           - Perl extension for consistent signal handl
p   libsys-statistics-linux-perl    - Perl module to collect system statistics  
p   libsys-syscall-perl             - module to access system calls that Perl do
p   libsys-syslog-perl              - Perl interface to the UNIX syslog(3) calls
p   libsys-syslog-perl:i386         - Perl interface to the UNIX syslog(3) calls
p   libsys-utmp-perl                - perl module to query UTMP files           
p   libsys-utmp-perl:i386           - perl module to query UTMP files           
p   libsys-virt-perl                - Perl module providing an extension for the
p   libsys-virt-perl:i386           - Perl module providing an extension for the
p   libsysadm-install-perl          - module for typical installation tasks of s
p   libsystem-command-perl          - Perl class for running system commands    
</pre>

axklim, 2016-04-11:
use command:
<pre>
sudo dpkg --ignore-depends=libsys-meminfo-perl -i zoneminder_1.28.108-nmu2015100101_amd64.deb
</pre>

zm install normaly

connortechnology, 2016-04-26:
https://www.debian-administration.org/article/78/Building_Debian_packages_of_Perl_modules

You can always build libsys-meminfo-perl yourself.

SteveGilvarry, 2016-07-27:
Closing this out as depends for packages is distro issue, and once this code is released I am sure the Debian maintainer will address in his package.

closed 2016-07-27 by SteveGilvarry