ZoneMinder/zoneminder #1415: Feature Request: remove apache2.2 from dependencies or reconfigure default script to run on custom port.

mrPsycho, 2016-04-12:
why? it seems that installing webserver with default configs forwards to common problems: 80 port is being used for other server.

so it would be great to remove apache from package or to reconfigure it to custom port.

SteveGilvarry, 2016-04-12:
Not a feature of ZoneMinder, dependencies are controlled by maintainers. For our own purposes we depend on apache2 | httpd, because we need one. But no apache config is automatically enabled so not really sure why you have an issue. Maybe your issue is more with the apache2 package automatically running on port 80, but that is not a ZM issue.
Equally the question is why is the other server using port 80, maybe it should be reconfigured?

closed 2016-04-12 by SteveGilvarry

knnniggett, 2016-04-12:
Boom.... @SteveGilvarry nailed it before I could comment.

@mrPsycho  In any case, if you need help with your configuration, head on over to the user forum, state your distro, how you installed zoneminder, and what exactly you are trying to accomplish.

mrPsycho, 2016-04-12:
@SteveGilvarry thanks!
now it seems clear.

and no, it shouldn't be reconfigured cause i have setup with nginx + php-fpm. 

ps. and yes, i reconfigured apache to run on different port.

SteveGilvarry, 2016-04-13:
It is possible to run zm under nginx Debian read me has some details, ask
in forum if you need help.