ZoneMinder/zoneminder #1417: Ability to clone a monitor

pliablepixels, 2016-04-14:
I'd like to propose a feature to ZM where the "Add Monitor" button could also serve as "cloning" a monitor. The idea is as follows: if you check an existing monitor, the "Add New Monitor" button changes to "Clone Monitor". Tapping the button in that state essentially copies all the configurations of that monitor into a new monitor setting. This also saves on screen real estate as opposed to having multiple buttons and dropdowns to clone. I'll submit a PR for review

connortechnology, 2016-04-14:
I disagree with the UI proposed.  Mainly because I want to get rid of those checkboxes.

I think opening up a monitor, there should be a copy button, which brings up a new popup...

pliablepixels, 2016-04-14:
those checkboxes are used to group delete

pliablepixels, 2016-04-14:
UX proposed: https://youtu.be/tn2NWHDiOQY

knnniggett, 2016-04-14:
I really like how you recycled the use of the "Add new monitor" button. This looks win-win to me.
Maybe we can get resolution to #1317 before we decide to merge this.

pliablepixels, 2016-04-14:
Not sure I see the problem described in #1317  - it works for me on both Chrome and Safari

connortechnology, 2016-04-14:
Yeah, how often you group delete?

How about dragndrop to a trashcan icon?

Isaac

connortechnology, 2016-04-14:
I don't like the button switch behaviour.  Now I can't add a monitor if I have a checkbox checked...
I'd rather see the checkboxes replaced with action buttons like copy, delete. Maybe with a pretty icon

pliablepixels, 2016-04-14:
how often will you create a new monitor with another checked ? 
And its not that you can't create - if checked, it clones. Want to add with a monitor checked? check another or uncheck

knnniggett, 2016-04-15:
So how about adding a Clone button, next to the Edit and Delete buttons, which only enables when 1 and only 1 monitor is checked?

knnniggett, 2016-04-20:
#1418 merged.

closed 2016-04-20 by knnniggett