ZoneMinder/zoneminder #1426: (FreeBSD) Default timestamp contains no monitor name

abishai, 2016-04-17:
Default timestamp value is %N - %d/%m/%y %H:%M:%S, it shows in stream as N - <timestamp>
If I add leading % (%%N - %d/%m/%y %H:%M:%S ) it starts to work properly. Not found where the code is to figure what's the problem.

josh4trunks, 2016-04-27:
is this set here?
https://github.com/ZoneMinder/ZoneMinder/blob/master/db/zm_create.sql.in#L353

josh4trunks, 2016-04-27:
looks like it's different than the default so I'm guessing not

josh4trunks, 2016-05-18:
Can anyone running linux using Zoneminder 1.29.1 comment? I'm wondering if this is a FreeBSD specific issue.

SteveGilvarry, 2016-05-18:
@josh4trunks FreeBSD only, here I think https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_monitor.cpp#L3142
[image]

josh4trunks, 2016-05-19:
Seems like a difference with BSD then. Like things running through this function need to be double escaped for some reason.

I confirmed '%f' also does not work, and needs the double **%**. Also position doesn't matter, %N / %f don't work properly anywhere in the label.

closed 2017-03-18 by SteveGilvarry