ZoneMinder/zoneminder #1430: ZM 1.29 can't get cURL source type to work

mdetweiler, 2016-04-21:
ZM 1.29 installed on Ubuntu 14.04 using the PPA

I have the exact same config working on 1.28.107 with serveral IP cameras from Samsung and Axis.

I thought there was an issue with using avcons instead of ffmpeg so I installed ffmpeg as below:
sudo add-apt-repository ppa:mc3man/trusty-media
apt-get update
apt-get dist-upgrade
apt-get install ffmpeg

I made sure to enable it in the config and set the path, but still get no image and no usable error mesage in syslog.

Any idea?

Here is a log excerpt as ZM starts:
Apr 20 21:39:17 system-z zmpkg[31065]: INF [Sanity checking States table...]
Apr 20 21:39:17 system-z zmpkg[31065]: INF [Command: start]
Apr 20 21:39:17 system-z zmdc[31076]: INF [Server starting at 16/04/20 21:39:17]
Apr 20 21:39:20 system-z zmpkg[31065]: INF [Starting up services]
Apr 20 21:39:20 system-z zmdc[31076]: INF ['zmc -m 1' starting at 16/04/20 21:39:20, pid = 31100]
Apr 20 21:39:20 system-z zmdc[31100]: INF ['zmc -m 1' started at 16/04/20 21:39:20]
Apr 20 21:39:20 system-z undef[31100]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:20 system-z zmdc[31076]: ERR ['zmc -m 1' exited abnormally, exit status 255]
Apr 20 21:39:20 system-z zmdc[31076]: INF [Starting pending process, zmc -m 1]
Apr 20 21:39:20 system-z zmdc[31105]: INF ['zmc -m 1' started at 16/04/20 21:39:20]
Apr 20 21:39:20 system-z zmdc[31076]: INF ['zmc -m 1' starting at 16/04/20 21:39:20, pid = 31105]
Apr 20 21:39:20 system-z undef[31105]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:20 system-z zmdc[31076]: INF ['zmc -m 2' starting at 16/04/20 21:39:20, pid = 31109]
Apr 20 21:39:20 system-z zmdc[31109]: INF ['zmc -m 2' started at 16/04/20 21:39:20]
Apr 20 21:39:20 system-z undef[31109]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:20 system-z zmdc[31076]: ERR ['zmc -m 2' exited abnormally, exit status 255]
Apr 20 21:39:20 system-z zmc[31105]: INF [Starting Capture version 1.29.0]
Apr 20 21:39:20 system-z zmdc[31076]: INF [Starting pending process, zmc -m 2]
Apr 20 21:39:20 system-z zmc[31105]: FAT [Unable to match Content-Type. Check URL, username and password]
Apr 20 21:39:20 system-z zmdc[31076]: ERR ['zmc -m 1' exited abnormally, exit status 255]
Apr 20 21:39:20 system-z zmdc[31076]: INF ['zmc -m 2' starting at 16/04/20 21:39:20, pid = 31113]
Apr 20 21:39:20 system-z zmdc[31113]: INF ['zmc -m 2' started at 16/04/20 21:39:20]
Apr 20 21:39:20 system-z zmdc[31115]: INF ['zmc -m 3' started at 16/04/20 21:39:20]
Apr 20 21:39:20 system-z zmdc[31076]: INF ['zmc -m 3' starting at 16/04/20 21:39:20, pid = 31115]
Apr 20 21:39:20 system-z undef[31113]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:20 system-z undef[31115]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:20 system-z zmdc[31076]: ERR ['zmc -m 3' exited abnormally, exit status 255]
Apr 20 21:39:20 system-z zmc[31113]: INF [Starting Capture version 1.29.0]
Apr 20 21:39:20 system-z zmdc[31076]: INF [Starting pending process, zmc -m 3]
Apr 20 21:39:20 system-z zmdc[31120]: INF ['zmc -m 3' started at 16/04/20 21:39:20]
Apr 20 21:39:20 system-z zmdc[31076]: INF ['zmc -m 3' starting at 16/04/20 21:39:20, pid = 31120]
Apr 20 21:39:20 system-z undef[31120]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:20 system-z zmdc[31123]: INF ['zmc -m 4' started at 16/04/20 21:39:20]
Apr 20 21:39:20 system-z zmdc[31076]: INF ['zmc -m 4' starting at 16/04/20 21:39:20, pid = 31123]
Apr 20 21:39:21 system-z zmc[31120]: INF [Starting Capture version 1.29.0]
Apr 20 21:39:21 system-z undef[31123]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:21 system-z zmdc[31076]: ERR ['zmc -m 4' exited abnormally, exit status 255]
Apr 20 21:39:21 system-z zmdc[31076]: INF [Starting pending process, zmc -m 4]
Apr 20 21:39:21 system-z zmdc[31076]: INF ['zmc -m 4' starting at 16/04/20 21:39:21, pid = 31130]
Apr 20 21:39:21 system-z zmdc[31130]: INF ['zmc -m 4' started at 16/04/20 21:39:21]
Apr 20 21:39:21 system-z undef[31130]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:21 system-z zmdc[31076]: INF ['zmc -m 5' starting at 16/04/20 21:39:21, pid = 31134]
Apr 20 21:39:21 system-z zmdc[31134]: INF ['zmc -m 5' started at 16/04/20 21:39:21]
Apr 20 21:39:21 system-z zmc[31130]: INF [Starting Capture version 1.29.0]
Apr 20 21:39:21 system-z zmc[31130]: FAT [Unable to match Content-Type. Check URL, username and password]
Apr 20 21:39:21 system-z zmdc[31076]: ERR ['zmc -m 4' exited abnormally, exit status 255]
Apr 20 21:39:21 system-z undef[31134]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:21 system-z zmdc[31076]: ERR ['zmc -m 5' exited abnormally, exit status 255]
Apr 20 21:39:21 system-z zmdc[31076]: INF [Starting pending process, zmc -m 5]
Apr 20 21:39:21 system-z zmdc[31141]: INF ['zmc -m 5' started at 16/04/20 21:39:21]
Apr 20 21:39:21 system-z zmdc[31076]: INF ['zmc -m 5' starting at 16/04/20 21:39:21, pid = 31141]
Apr 20 21:39:21 system-z undef[31141]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:21 system-z zmdc[31076]: INF ['zmc -m 6' starting at 16/04/20 21:39:21, pid = 31144]
Apr 20 21:39:21 system-z zmdc[31144]: INF ['zmc -m 6' started at 16/04/20 21:39:21]
Apr 20 21:39:21 system-z undef[31144]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
[... 126 lines cut ...]
Apr 20 21:39:56 system-z zmdc[31261]: INF ['zmc -m 5' started at 16/04/20 21:39:56]
Apr 20 21:39:56 system-z zmdc[31076]: INF [Starting pending process, zmc -m 4]
Apr 20 21:39:56 system-z zmdc[31076]: INF ['zmc -m 4' starting at 16/04/20 21:39:56, pid = 31263]
Apr 20 21:39:56 system-z zmdc[31263]: INF ['zmc -m 4' started at 16/04/20 21:39:56]
Apr 20 21:39:56 system-z zmdc[31076]: INF [Starting pending process, zmc -m 6]
Apr 20 21:39:56 system-z undef[31261]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:56 system-z zmdc[31076]: INF ['zmc -m 6' starting at 16/04/20 21:39:56, pid = 31265]
Apr 20 21:39:56 system-z zmdc[31265]: INF ['zmc -m 6' started at 16/04/20 21:39:56]
Apr 20 21:39:56 system-z undef[31263]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:56 system-z zmc[31261]: INF [Starting Capture version 1.29.0]
Apr 20 21:39:56 system-z zmc[31261]: FAT [Unable to match Content-Type. Check URL, username and password]
Apr 20 21:39:56 system-z zmdc[31076]: ERR ['zmc -m 5' exited abnormally, exit status 255]
Apr 20 21:39:56 system-z undef[31265]: INF [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
Apr 20 21:39:56 system-z zmc[31263]: INF [Starting Capture version 1.29.0]
Apr 20 21:39:56 system-z zmc[31263]: FAT [Unable to match Content-Type. Check URL, username and password]
Apr 20 21:39:56 system-z zmdc[31076]: ERR ['zmc -m 4' exited abnormally, exit status 255]
Apr 20 21:39:56 system-z zmc[31265]: INF [Starting Capture version 1.29.0]
Apr 20 21:39:56 system-z zmc[31265]: FAT [Unable to match Content-Type. Check URL, username and password]
Apr 20 21:39:56 system-z zmdc[31076]: ERR ['zmc -m 6' exited abnormally, exit status 255]

The monitors that work fine are, but are not using source type cURL:
2,3,7,8,9,10

The monitors that are having trouble and using source type cURL:
1,4,5,6

mdetweiler, 2016-04-22:
Had some brief hope that I found the issue when I noticed that my working older install has the package below and some old threads about geting cURL to work say you need the -dev package.

apt-get install libcurl4-gnutls-dev

No luck still.

knnniggett, 2016-04-22:
We have previously asked you to follow our posting rules:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

The issue you have created in our github forum is not a bug report but is a request for help configuring your cameras. 

We will be glad to help you configure cameras in the user forum, in the thread you already created there: https://forums.zoneminder.com/viewtopic.php?f=34&t=24599

closed 2016-04-22 by knnniggett