ZoneMinder/zoneminder #1431: Issues with image rotation

tommycw1, 2016-04-22:
I have a 16:9 camera that shoots into a narrow driveway, so I have the camera physically rotated right to make the most of the field of view.

In my zm source setup up to 1.29.0 I had the "rotate right" option set and I transpose the normal 1920x1080 resolution to 1080x1920.  

Today I finally upgraded to the latest 1.29.1.  When I pulled up the camera live view I noticed the image was squashed (deformed) to 1920x1080, so I went into my source setting and changed the resolution back from 1080x1920 to 1920x1080.  Pulled up the live view and problem solved.  Sounded good to me, and I think I even remembered reading some change in ZM that made it so you didn't have to manually adjust the resolution of a source when rotating.  Great!

Problem I have is when I bring up the zones view, the image I get is a 1080x1920 image squished (deformed) to 1920x1080.  Same problem in viewing events from this camera, the Montage View, and in the 3rd party app ZMNinja.  Seems that the only place that auto rotates the resolution correctly is the live camera view.  No matter now I set this I cannot get the live view to match the other views.

I am running Ubuntu 15.10 previous version that worked was 1.29.0 and broken version is 1.29.1

[image]
[image]

pliablepixels, 2016-04-24:
With respect to zmNinja, my guess is this change has probably affected things:
https://github.com/ZoneMinder/ZoneMinder/commit/45fc689203b9051cc95574956b0802c11b13b092

@tommycw1  - I don't have a vertically rotated camera to test with this sort of aspect ratio and doing it on a regular camera is confusing me on what the behavior should be.  Let me know if you can give me access to this camera via a test user account for a day and I'll try and fix it for zmNinja

pliablepixels, 2016-04-24:
@connortechnology - I am not sure the current implementation of the rotation is correct in 1.29. I've implemented the same logic in zmN and it now acts like ZM does, but this behavior seems incorrect. What we are essentially doing is specifying the monitor size as "X*Y" while the image rotated really is producing "Y*X" and then we are force changing width and height in the image tag by swapping w/h --> this produces incorrect stretching - evident from the timestamp on top (in this image posted as well as my test with a new ZM build with a camera laying on the side)

[image]

tommycw1, 2016-04-25:
@pliablepixels - sure I can give you access.  What permissions do you need, just viewing of that monitor or will you need full control?  What is the best way to give you the url/uname/passwd without making it public here?

pliablepixels, 2016-04-25:
@tommycw1  - please shoot me an email - pliablepixels@gmail - just view is fine

pliablepixels, 2016-04-25:
@tommycw1   - I managed to duplicate it by rotating one of my cameras, so I don't need access. Thanks for the offer.

tommycw1, 2016-04-25:
OK, thanks for your work on this.  Let me know if you need anything else
from me.  More than happy to help out.

pliablepixels, 2016-04-26:
FYI this PR (work in progress) by @connortechnology fixes this at the ZM end - no fixes needed in zmNinja. Once this PR is fully tested and merged to master and released ,zmNinja will automatically start showing it correctly

tommycw1, 2016-04-26:
Great, thanks to all for the super fast fix!

I'm guessing this will be included in 1.29.2?  What is the normal time for
a new release to come out?  Not sure if you guys do releases on a set time
frame or just whenever you collect enough changes.

Thanks again,

Tom

SteveGilvarry, 2016-07-27:
Fixed #1442

closed 2016-07-27 by SteveGilvarry