ZoneMinder/zoneminder #1435: Build Package From Source video-storage

aljabr, 2016-04-23:
hi,
i get tired trying to Build The Package From Source 
but nothing work

i think there is lack of information 
is there anyone can help us and update the documentation for centos 7
thank you

aljabr, 2016-04-23:
how can i post a bounty for this issue? :)

SteveGilvarry, 2016-04-23:
Please read our posting rules, then feel free to seek help in our user forum:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

Supporting development branch builds is not something I am going to do in here, they are experimental by their nature and those who need to run them are able to. This will make it's way to next zoneminder release and you will be able to install from zmrepo.

closed 2016-04-23 by SteveGilvarry