ZoneMinder/zoneminder #1436: (FreeBSD) Not all optional header being detected by cmake

abishai, 2016-04-23:
Video4linux is not detected as checks lacks proper include path

Determining if the include file libv4l1-videodev.h exists failed with the following output:
Change Dir: /usr/home/abishai/zoneminder/work/ZoneMinder-5a3978f/CMakeFiles/CMakeTmp

https://gist.github.com/abishai/efc6ef0b452de244f60f64bf44b7d7b3

abishai@freezm:~/zoneminder % ls -la /usr/local/include/ | grep libv4l1-videodev.h
-rw-r--r--   1 root  wheel   7135 Apr 23 23:13 libv4l1-videodev.h 

You may see, this directory is not included as a search path.

abishai, 2016-04-24:
Will be solved in port config
-DCMAKE_REQUIRED_INCLUDES:STRING="${LOCALBASE}/include" should be added, as BSD keeps include stuff in /usr/local

closed 2016-04-24 by abishai

knnniggett, 2016-04-24:
The compiler is failing to find this library, rather than cmake.
If cmake had not found it then it would not have been included in the build because v4l is optional. You did not show your entire build output (please do from now on), but if you look at the beginning you will likely see cmake finding that header file.

I had a similar problem with ffmpeg, primarily on redhat builds. Ffempeg likes to stick their header files into subfolders, like libavformat, which caused a similar build problem.

The solution I liked best was to do something like this: 
https://github.com/ZoneMinder/ZoneMinder/blob/master/CMakeLists.txt#L413

If you like the solution you've already got then we can add that parameter to the FreeBSd target distro flag.

abishai, 2016-04-24:
That gist is for cmake checks, it created dummy file and try to resolve header files, so this is not compiler one. 
-- Looking for libv4l1-videodev.h
-- Looking for libv4l1-videodev.h - not found
-- Looking for linux/videodev.h
-- Looking for linux/videodev.h - not found
-- Looking for linux/videodev2.h
-- Looking for linux/videodev2.h - not found
Build succedes, but without v4l support, that's why I included only part of log. Passing the CMAKE_REQUIRED_INCLUDES resolved the issue.

However, the question I have (and I can't check it myself as I don't have suitable hardware) - CMAKE_REQUIRED_INCLUDES affecting only check_include_file, so is v4l support was actually compiled on? I suspect, the answer is yes, as failure to detect header during compilation would stop the build process.