ZoneMinder/zoneminder #1437: Continue to build ffmpeg from source in travis?

kylejohnson, 2016-04-24:
Our [.travis.yml](https://github.com/ZoneMinder/ZoneMinder/blob/master/.travis.yml#L28-L34) file has us building ffmpeg from source.  Should we continue to do so, or should we just install it from the apt repo?

The latter would cut down build times quite a bit, and could keep us at a stable / supported ffmpeg version.

SteveGilvarry, 2016-04-24:
Travis is very slow to update the build servers so until a couple of months ago we would have been stuck on Ubuntu 12.04 associated ancient ffmpeg/libav. Now just as the newest LTS version is released we are now on the previous LTS. We are no longer building off ffmpeg master, so currently 3.0 which is the current release.
So my vote is yes, do build times matter that much in Travis?

knnniggett, 2016-04-24:
I'm not too concerned with build time myself. Since ffmpeg changes so frequently, I find value in building zoneminder against very new versions of ffmpeg to spot troubles before that version of ffmpeg finds its way into a Linux distro.

SteveGilvarry, 2016-04-25:
There is another option, since we are now on 14.04 for the builds I can try the PPA version that ffmpeg link to on their site for trusty. I will test it out on my repo that is linked to travis separately. But I actually think onvif being so verbose is more of a problem.

SteveGilvarry, 2016-04-26:
No decent official or close to offical PPA for trusty that has the shared libraries.

SteveGilvarry, 2016-07-24:
I halved build time in travis by using caching

closed 2016-07-24 by SteveGilvarry