ZoneMinder/zoneminder #1440: Invalid datetime format when saving monitor settings

dulinux, 2016-04-24:
Hello, fresh zm install on ubuntu 16.04 with mariadb, following zm wiki instructions. At first, everything went just great. I was actually recording events already. Then I needed to shutdown the machine and put it at another place. After startig it again, I'm getting this when trying to add or modify any monitor setting:

`2016-04-24 20:38:06.388063	web_php		2743	FAT	SQL-ERR 'SQLSTATE[22007]: Invalid datetime format: 1366 Incorrect integer value: '' for column 'ServerId' at row 1', statement was 'update Monitors set ServerId = '', AnalysisFPS = '', MaxFPS = '', AlarmMaxFPS = '', Host = '192.168.1.50', Port = '80', Triggers = '' where Id =?'	/usr/share/zoneminder/www/includes/database.php	115
`

Thanks a lot.

SteveGilvarry, 2016-04-25:
Mysql 5.7 related issue in how it handles NULL and '', we are wrong, but have to deal with 3 different db access methods between C++, PHP and Perl.
Can be worked around using sql mode settings as shown in the wiki, maybe just check you got that setting correct. 
https://wiki.zoneminder.com/Ubuntu_Server_16.04_64-bit_with_Zoneminder_1.29.0_the_easy_way

dulinux, 2016-04-26:
I had STRICT_TRANS_TABLES in the sql-mode setting. Removing it apparently made things work again. Tks

closed 2017-03-18 by SteveGilvarry