ZoneMinder/zoneminder #1451: Can't enable monitor from the API.

stenius, 2016-04-30:
When I make a request to `/zm/api/monitors/1.json` with `{'Monitor[Function]': 'Modect', 'Monitor[Enabled]': 'false'}` I can disable an enabled monitor, however when I try to POST `{'Monitor[Function]': 'Modect', 'Monitor[Enabled]': 'true'}` to a disabled monitor, I can't enable it.

I am running version 1.29.0 under Ubuntu 14.04.

pliablepixels, 2016-04-30:
Works for me

Initial monitors:
```
ps -aef | grep zmc
www-data  4677  4648  5 08:05 ?        00:01:43 /usr/bin/zmc -m 1
www-data  4701  4648  1 08:05 ?        00:00:25 /usr/bin/zmc -m 3
www-data  4715  4648  3 08:05 ?        00:00:55 /usr/bin/zmc -m 4
www-data  4723  4648  3 08:05 ?        00:00:57 /usr/bin/zmc -m 5
www-data  4737  4648  6 08:05 ?        00:02:02 /usr/bin/zmc -m 6
www-data  5749  4648  4 08:35 ?        00:00:00 /usr/bin/zmc -m 2
```

Disable monitor 2:
```
curl -XPUT https://server/zm/api/monitors/2.json -d "Monior[Function]=None&Monitor[Enabled]:false" -b cookies.txt -k
```

Check to see monitor is disabled and zmc is killed
```
ps -aef | grep zmc
www-data  4677  4648  5 08:05 ?        00:01:45 /usr/bin/zmc -m 1
www-data  4701  4648  1 08:05 ?        00:00:25 /usr/bin/zmc -m 3
www-data  4715  4648  3 08:05 ?        00:00:56 /usr/bin/zmc -m 4
www-data  4723  4648  3 08:05 ?        00:00:57 /usr/bin/zmc -m 5
www-data  4737  4648  6 08:05 ?        00:02:04 /usr/bin/zmc -m 6
```

Now re-enable

```
curl -XPUT https://server/zm/api/monitors/2.json -d "Monior[Function]=Modect&Monitor[Enabled]:true" -b cookies.txt -k
```

Check to see if its up
```
ps -aef | grep zmc
www-data  4677  4648  5 08:05 ?        00:01:46 /usr/bin/zmc -m 1
www-data  4701  4648  1 08:05 ?        00:00:25 /usr/bin/zmc -m 3
www-data  4715  4648  3 08:05 ?        00:00:57 /usr/bin/zmc -m 4
www-data  4723  4648  3 08:05 ?        00:00:58 /usr/bin/zmc -m 5
www-data  4737  4648  6 08:05 ?        00:02:06 /usr/bin/zmc -m 6
www-data  5779  4648 27 08:36 ?        00:00:00 /usr/bin/zmc -m 2
```

stenius, 2016-04-30:
So that works for me, using POST instead of PUT, but what I am doing differently is **not** changing Monitor[Function] to None while disabling so the monitoring process stays up.

From the web UI I can open a monitor and enable/disable it without changing the function.

connortechnology, 2016-05-07:
I think this is because on edit, we pass the request data to the daemonControl function as a monitor object/array, but it isn't complete, so the daemonControl function fails. I have a PR. With my change, I have successfully disabled and re-enabled. It seem to take a couple of seconds for zma to start back up,

PR #1467 (2016-05-07): Don't pass request data as a monitor array to daemonControl
fixes #1451 I think
closed 2016-05-07 by connortechnology