ZoneMinder/zoneminder #1461: Logic of Event Close Mode is flawed

connortechnology, 2016-05-05:
If you have a monitor that is always alarmed (perhaps due to oversensitive zones) you end up with events that last forever.  This causes other problem when trying to view them.

I propose that in mocord mode, we always honour the section length setting, using it as a maximum event length.  I propose that we also close events on change of state, from idle->alarm and vice versa.  

And then we do away with the setting altogether.

Thoughts?

pliablepixels, 2016-05-13:
agreed - segment lengths should be honored - APIs may overflow too

chrisc1073, 2017-12-14:
Honouring a maximum Section length setting for Modect is a must, is generally a standard and would be fantastic.  I have several cameras in crowded rooms and have to wait hours (sometimes all day) for an event to return to idle, register on the main page, so I can then select it and see it.  This is no good if you need to quickly review some footage of an incident that occurred in the last 10 mins.

connortechnology, 2017-12-14:
It occurs to me that a force end button would be useful in that circumstance too.

chrisc1073, 2017-12-14:
Yes great idea if you need to see something that just happened urgently.  In fact, I was thinking that given that all the new images actually exist on the hard disk, maybe there just needs to be a quick way to shuffle through them as part of any "new (but unregistered) event".

pliablepixels, 2017-12-14:
From a UI perspective, one could just show the frames recorded so far - that is what I do in zmNinja - I had the same problem. So I just display what is already available. However, in the case of video, I suppose depending on how the video is being encoded, this may or may not work.

digital-gnome, 2017-12-14:
On 1.31 using jpegs record and mocord display whatever has been recorded so far if the monitor is idle even if the event has not been closed yet.  Oddly that doesn't work if the monitor is alarmed, don't see any reason for that.  

It's possible to force an end by using disable alarms on the watch view.  A quick toggle shouldn't lose much footage from a modect monitor.

Passthrough is a whole different ball of wax.  Disable alarms would still end an event but viewing in progress mocord/record would take some work.

chrisc1073, 2017-12-14:
To overcome the issue of modect never stopping in crowded areas I run a script and execute it via cron every 10 mins.  It will just kill all analysis but then it will restart itself straight away.  Its brutal but it works. 
----------------------
#!/bin/bash
# script to kill all zoneminder analysis
killall -v zma
----------------------

Alternatively use several of these below instead of killall to target specific monitor ids
pkill -e -x -f "usr/bin/zma -m 6"
pkill -e -x -f "usr/bin/zma -m 10"
etc

closed 2018-06-22 by stale[bot]