ZoneMinder/zoneminder #1477: OpenBSD cmake condition to skip librt

czettnersandor, 2016-05-12:
Because in OpenBSD, clock_gettime is provided by libc and librt does not exist, I came up with this updated line in CMakeLists.txt

if (UNIX AND NOT ${CMAKE_SYSTEM_NAME} MATCHES "OpenBSD")

Is this the proper solution? I'm not familiar with cmake, but it's a solution on OpenBSD, cmake is making further progress after the change. I'll check if this is the case in FreeBSD and other BSD like systems and add it to the condition.

josh4trunks, 2016-05-13:
The updated FreeBSD port is being worked on here.
https://github.com/abishai/Zoneminder-port/tree/master/files

There are some patched to CMakeLists.txt so assuming OpenBSD uses a similar ports system you can probably do a similar thing.

czettnersandor, 2016-05-13:
Great thanks, at least I will not work twice. Interesting He didn't disable librt, probably it's only a problem in OpenBSD.

closed 2017-03-18 by SteveGilvarry