ZoneMinder/zoneminder #1481: Debian 1.29 backport repo version: Incomplete.

digdilem, 2016-05-15:
Bug report that debian doesn't seem to have a clean zoneminder package since Jessie.

Not in standard repos (1.24.2 was in Wheezy, which was the last time I installed zm... :)  )

A quick google said 1.29 was in the official backports repo (well done, whoever did that)

However, installation was less than trivial and ran into a number of fairly familiar problems, so I am reporting them here as one man's experience of installing it. (It's still a lot easier than the early versions. I recall taking a week to build one of the very early public releases)

If this is deemed to be a packaging bug rather than a zoneminder one, please advise where better to report it. Thanks.

=====

Brand new install of debian-stable yesterday onto bare metal.

deb http://http.debian.net/debian jessie-backports main
Provides:  zoneminder-dbg,zoneminder 1.29.0+dfsg-1~bpo8+1

Zoneminder's config was successfully added to /etc/apache/conf-available but was not enabled.

No zm database created. (I manually ran zm_create.sql)

May 15 09:32:56 dobby zmpkg.pl[2198]: DBI connect('database=zm;host=localhost','zmuser',...) failed: Access denied for user 'zmuser'@'localhost' (using password: YES) at /usr/share/perl5/ZoneMinder/Config.pm line 92.

No zmuser added with privs to db, manually added.

May 15 09:42:25 dobby zmpkg.pl[4332]: Can't open config file '/etc/zm/zm.conf': Permission denied at /usr/share/perl5/ZoneMinder/Config.pm line 119.

chmod 755 to that file allowed zm to start. 

Webconsole then shows; "ZoneMinder is not installed properly: php's date.timezone is not set to a valid timezone"

Manually fixed that.

Then: 

Webconsole: "Cannot write to content dirs('events','images'). Check that these exist and are owned by the web account user"

chown www-data.www-data for /usr/cache/zoneminder -R

Then console was displayed.

I added an ip camera, using the same details from a working zm 1.25 install. Monitor went link-green, but no video showing.

A quick check of the filesystem shows that the jpgs are being created normally.

Apache's error.log has these, presumably linked to the form-submission when I saved the monitor.
[Sun May 15 09:53:22.605970 2016] [:error] [pid 5612] [client 10.1.0.150:58151] PHP Notice:  Undefined index: User in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 556, referer: http://10.1.5.11/zm/index.php
[Sun May 15 09:53:22.606024 2016] [:error] [pid 5612] [client 10.1.0.150:58151] PHP Notice:  Undefined index: Pass in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 557, referer: http://10.1.5.11/zm/index.php
[Sun May 15 09:53:22.606049 2016] [:error] [pid 5612] [client 10.1.0.150:58151] PHP Notice:  Undefined index: RTSPDescribe in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 573, referer: http://10.1.5.11/zm/index.php
[Sun May 15 09:53:55.443085 2016] [:error] [pid 5595] [client 10.1.0.150:58285] PHP Notice:  Undefined index: User in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 556, referer: http://10.1.5.11/zm/index.php
[Sun May 15 09:53:55.443134 2016] [:error] [pid 5595] [client 10.1.0.150:58285] PHP Notice:  Undefined index: Pass in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 557, referer: http://10.1.5.11/zm/index.php
[Sun May 15 09:53:55.443158 2016] [:error] [pid 5595] [client 10.1.0.150:58285] PHP Notice:  Undefined index: RTSPDescribe in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 573, referer: http://10.1.5.11/zm/index.php

Streaming shows these;
[Sun May 15 09:59:10.307653 2016] [:error] [pid 5612] [client 10.1.0.150:59584] ERR [socket_bind( /var/run/zm/zms-040843w.sock ) failed: Address already in use], referer: http://10.1.5.11/zm/?view=watch&mid=1
[Sun May 15 09:59:10.937750 2016] [:error] [pid 5595] [client 10.1.0.150:59565] ERR [socket_sendto( /var/run/zm/zms-040843s.sock ) failed: No such file or directory], referer: http://10.1.5.11/zm/?view=watch&mid=1

There is no zms-*.sock file present, nor in /tmp/zm/

"a2enmod cgi && apache2ctl restart"  fixed that. Apache's cgi module wasn't enabled.

This ubuntu guide from January appears to provide most of the resolution steps, suggesting the package wasn't perfect there, either?

https://wiki.zoneminder.com/Ubuntu_Server_15.04_or_15.10_64-bit_with_Zoneminder_1.29.0_the_easy_way

digdilem, 2016-05-15:
Please note this is distinct from the other debian bug report, #1413, as that involves building from source.

SteveGilvarry, 2016-05-15:
More a packaging thing, with so many components and potential different setups, sometimes things are left for the user to configure as they like. If you feel some of your issues are bugs then maybe try talking with the packager. But he will likely refer you to the README.Debian that is part of the package.
https://anonscm.debian.org/cgit/collab-maint/zoneminder.git/tree/debian/README.Debian?h=jessie-backports

digdilem, 2016-05-15:
Thanks for pointing out that file, I had overlooked it and it makes clear that all my issues are already known to the package maintainer, so no need for this bug to stay active.

closed 2016-05-15 by digdilem