ZoneMinder/zoneminder #1482: Do we need the multi-server "Not using multi server" logs?

pliablepixels, 2016-05-15:
Specifically,
https://github.com/ZoneMinder/ZoneMinder/blob/66a3fef61ebd10c45968a3706021d45d027e28be/src/zm_config.cpp#L138-L143

It seems when any operation results in some access to loadConfig, this log is generated. In R1.29 this was an INFO log, which we have demoted to Debug. We can keep it in Debug, but do note that for any operation, such as cgi-bin/zms view snapshot (or jpeg),any API that does a loadConfig (like alarmStatus) will keep generating this log. Folks using zmNinja with the next release of ZM (albeit in debug mode) will be rather overwhelmed as I am using alarm status APIs extensively.

@connortechnology thoughts?

knnniggett, 2016-05-15:
I was thinking we should move this event to display once, during zoneminder startup only.

pliablepixels, 2016-05-15:
Yes, I do think that's a more appropriate place. ``zmLoadConfig`` is called by all the processes during their inits, (zma/zms/zmu/zmc) - at the C++ level I don't know what would constitute a zoneminder startup beyond initing these processes. So even if this check is done after each zmLoadConfig in all or one of these processes, that should suffice too. Of these processes, I think zms is the one that restarts the most.

PR #1499 (2016-05-24): move multiserver message to display once during startup only
fixes #1482 

It should work, but I haven't tested this yet.
closed 2016-05-24 by connortechnology