ZoneMinder/zoneminder #1483: Show event disk usage stats

connortechnology, 2016-05-16:
When listing events, I think I'd like to know how much space a given event is taking up.  

We could add this up while recording, or have it done later in a background task.  Ideally I would like it stored in the db instead of doing a du everytime the events list is loaded.

mattdurant, 2016-09-01:
Are you thinking of something as simple as adding a FileSize column to the Frames table and altering the INSERT statement to insert the size off the Image object?  Should be easy to select the sum of the frame sizes to get the event size on disk at that point for display.

connortechnology, 2016-09-01:
Not really.  It might be useful to have the frame size in the db, but an event can consist of frame images, mp4 video, thumbnails, etc.  I think having a column for event disk space and having it get updates whenever new event content is generated will be best.  The zmaudit script can double check it.

mattdurant, 2016-09-01:
The Event deconstrutor is already responsible for updating the Events table with final information, we could either run du and add this to the query there or trigger one of the perl scripts to do this so we're not holding up the deconstructor.  Another "low hanging fruit" option would be to leave it unpopulated and set up a background filter to do it every minute or so.  I like that better than leaving it all up to zmaudit as the gap where the info is missing would be smaller.  I like having zmaudit double check it for accuracy in case someone deletes frames, exports a video, etc to increase the size later.

connortechnology, 2016-09-01:
In my storageareas branch, when listing events, I have it doing a du on the event dir.  Performance is not bad.  If the size column is NULL, we could do a du.  We could also update it when viewing a specific event.

mattdurant, 2016-09-01:
I would think a combination of having zmaudit update nulls when it encounters them, and updating nulls when browsing would be good enough.  If the zmaudit hasn't updated the table, you'd only incur the performance hit once on the first view, and you should never have an instance of viewing events and not seeing the size.

The only issue I see with this approach is that if you want to do any filtering logic using this data, you run the risk of being inaccurate because of the small window of events having a null size.  Is the goal just for viewing or would you have this be used for keeping storage use down like we do know with the PurgeWhenFull filter?

jamiemaher, 2016-10-26:
This would be a good feature that could also help to keep storage use down.

If an event's size was known it would be easier to manually prune large false positives.

Or it could also be a first step in working towards an adaptive purge when full filter? Used in conjunction with keeping track of the average number of bytes zm wrote over the last minute or so.  The purge filter could then select a variable number of the oldest events to delete. For example matching a number of events to delete that sum up to more than the number of bytes written.