ZoneMinder/zoneminder #1485: Release 1.30.0

knnniggett, 2016-05-16:
This issue has been created as a placeholder to document what we need to address before our next release.

Due to some of the changes we've implemented, I plan to create a draft release before releasing. I estimate the draft release process to take approx. 1 month.

@ ZoneMinder Team
- Personally, I'm fine with calling our next release 1.29.1, instead of 1.30.0, but I'll go with popular opinion. Please voice your opinion on what we want to call our next release.
- We have a lot of open pull requests. Review and discuss which ones should be part of our next release

@kylejohnson 
- Communicate to the rest of the team the state of the telemetry server. We want to make sure it is running reliably before we release. If there is anything needed, such as hardware upgrades, to minimize risk of downtime then let's have that discussion before we release. 

@knnniggett
- Review and test #1411 

@connortechnology @knnniggett
- discuss and implement a solution to #1482

connortechnology, 2016-05-16:
I think 1.30 is the correct version  number.  It has onvif turned on, and telemetry.  These are new features.  1.29.1 would be a bug fix release.

We need #1486 in to make @pliablepixels happy.

Also #1478 to fix some multi-server problems.

#1470 makes configuring the api a lot simpler.

josh4trunks, 2016-05-16:
I'm hoping we get the FreeBSD port/pkg updated with the release of 1.30!

https://github.com/abishai/Zoneminder-port is working built off `ff3e9c8`, but I noticed some functions stopped working when I updated the port to `9377dca`. See https://github.com/abishai/Zoneminder-port/issues/4#issuecomment-218384770

I'll test again on the latest commit and post a bug report if it's still not working.

connortechnology, 2016-05-17:
We should probably fix #1464 too.

knnniggett, 2016-05-18:
Well, there is something still messed up with the master branch. I'm placing this hear for now until I've got time to troubleshoot this further:
```
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/vendor_perl/ZoneMinder/Logger.pm line 156.
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/vendor_perl/ZoneMinder/Logger.pm line 206.
Use of uninitialized value $ZoneMinder::Config::Config{"ZM_DB_HOST"} in pattern match (m//) at /usr/share/perl5/vendor_perl/ZoneMinder/Database.pm line 83.
Use of uninitialized value $ZoneMinder::Config::Config{"ZM_DB_NAME"} in concatenation (.) or string at /usr/share/perl5/vendor_perl/ZoneMinder/Database.pm line 100.
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/vendor_perl/ZoneMinder/Database.pm line 100.
DBI connect('database=;host=','',...) failed: Access denied for user 'root'@'localhost' (using password: NO) at /usr/share/perl5/vendor_perl/ZoneMinder/Database.pm line 100.
Can't call method "trace" on an undefined value at /usr/share/perl5/vendor_perl/ZoneMinder/Database.pm line 102.
```
This is from one of my weekly builds on Fedora.

josh4trunks, 2016-05-18:
@knnniggett what commit did you build off? Database connections are working for me with b4b9c1f

knnniggett, 2016-05-18:
b4b9c1f 

Definitely broken in Fedora and CentOS. Weekly builds.
Commit 6e32b5d changed the way the Config hash is referenced.

josh4trunks, 2016-05-18:
Strange, mine is working, as a fresh install on FreeBSD. Maybe because I already https://github.com/ZoneMinder/ZoneMinder/pull/1492?
But that doesn't change that first reference to **$ZoneMinder::Config::Config{"ZM_DB_HOST"}**

connortechnology, 2016-05-18:
which .pl is generating that output?

connortechnology, 2016-05-18:
basically the config hasn't been initialized yet... meaning that ZoneMinder.pm hasn't been included yet.

knnniggett, 2016-05-18:
That output shows when I try to start zoneminder from the command line.
So it's either in zmpkg.pl or zmdc.pl

knnniggett, 2016-05-18:
Just a note that this user encountered the same problem a day after 6e32b5d was merged:
https://forums.zoneminder.com/viewtopic.php?f=34&t=24676&p=94589

At the time of my first response in that thread, I was unaware of the recent changes to database.pm.

pliablepixels, 2016-05-20:
Are we at code/issue freeze for 1.30? If not, there are a couple of benign changes (API documentation and a fix for #1493 for consideration. Neither affect zmNinja, so its your call.

knnniggett, 2016-05-20:
Right now the master branch still isn't working in all cases due to recent changes to the Perl code while I was on vacation. 

There will always be items we "need" to add to the source code, and right now I'm not willing to entertain anything other than fix the master branch and resolve what is already in this thread.

abishai, 2016-05-23:
I'd like to test master under FreeBSD before 1.30.0 release to use release branch for port.
Can you ping me when it will be almost ready ?

SteveGilvarry, 2016-05-26:
@knnniggett nothing for me, wasted about a week on spammers

knnniggett, 2016-06-01:
Release Candidate 1 is now official:
https://github.com/ZoneMinder/ZoneMinder/releases/tag/v1.30.0-rc1

Let the package building marathon begin.

connortechnology, 2016-06-01:
Thanks @knnniggett ! Um. I think we still need to bump the version files though.

knnniggett, 2016-06-01:
Not sure which file you mean. The one that matters has been bumped to 1.30.0:
https://github.com/ZoneMinder/ZoneMinder/blob/master/CMakeLists.txt#L7

Along with the corresponding update script:
https://github.com/ZoneMinder/ZoneMinder/blob/master/db/zm_update-1.30.0.sql

If you are referring to the file "version" in the root folder, then no I did not bump that. That was intentional, due to the previous mass confusion from our user base caused by bumping that file too early. It'll get bumped when the release is official and not just a candidate.

knnniggett, 2016-06-01:
@connortechnology I don't know if you did it on purpose, but you just deleted the 1.30.0 source tarball from our releases page.

connortechnology, 2016-06-01:
Sorry...

The file that has to be changed is called version.

connortechnology, 2016-06-01:
okay, I give up.

connortechnology, 2016-06-01:
packaging uses that file.  I don't think pre-releases should have a tarball called 1.30.  It should 1.30.rcwhatever.  I also don't think we should be checking master branch's version file to see what the latest release is.  Because it is not released.  We need to fix this confusion.

knnniggett, 2016-06-01:
I don't disagree with you, but @kylejohnson controls how the version file is used. Only he can change it as far as I know. 

The tarball needs to be named that way because rpm packaging scripts require it to have that name.

connortechnology, 2016-06-01:
Let's get this fixed before we go further.  I will bug Kyle.

knnniggett, 2016-06-01:
Now, I am going to step away from my pc today and celebrate my wife's 40th birthday today. Please don't break anything while I am away.

connortechnology, 2016-06-01:
I will not.  Enjoy your day and hers!

SteveGilvarry, 2016-06-01:
Just don't bump the one used by version check until actual release, or all
and sundry will start installing rc1.

pliablepixels, 2016-06-01:
@SteveGilvarry - don't be a spoil sport. Let's have some mayhem, shall we 👍

connortechnology, 2016-06-01:
Let's have some proposals for fixing this mess.  

Code that looks for the latest version should look for the most recent release that doesn't have an rc on the end of it.  It should NOT look at a version file in master branch.

connortechnology, 2016-06-01:
Apparently wget https://api.github.com/repos/ZoneMinder/ZoneMinder/releases/latest will give us a nice json listing of the most recent release.

pliablepixels, 2016-06-01:
right, that is what[ I use in zmN](https://github.com/pliablepixels/zmNinja/blob/master/www/js/app.js#L665-L683) - I was about to post that. You can simply check for the presence of "rc" there and if so, not take it as an update. I don't think there is any need for a separate version file

connortechnology, 2016-06-03:
Anyone know how/where https://update.zoneminder.com/version.txt comes from?

knnniggett, 2016-06-03:
I thought @kylejohnson went with my suggestion and wrote a cron script, which would look for a change in the "version" file on github and then update "vesion.txt" accordingly on update.zoneminder.com.

I'm not sure I understand the scrutiny going on here. That file is currently working exactly as it should. It needs to stay at 1.29.0 until 1.30.0 becomes an official release.

pliablepixels, 2016-06-03:
I think that version file is not really necessary and adds an extra layer of manual intervention (and delay). Given github APIs achieve this automatically each time a new release is uploaded and is able to classify what is a release vs. pre-release, it might be a better approach.

connortechnology, 2016-06-03:
Because it is NOT working.  It does not make sense to have that file be out of date, meaning not contain the same version value as the rest of the code.  Because I use it when I build packages, so at the moment, I am not making packages that say 1.30.  

whatever cron script generates that file, should look at our releases, not at master branch.

knnniggett, 2016-06-03:
> Because I use it when I build packages, so at the moment, I am not making packages that say 1.30.

@connortechnology Why didn't you say that 4 posts ago? That file was never intended to be used that way, and nobody but you knew you were using it in that manner. 

The version file has been out of sync since we bumped the rest of the code to 1.29.1, which was months ago.

In the short term, can't you just patch the verson file? I'm sure debian packaging scripts allow one to apply patches to the build. Alternatively, we could bump the version file in the release-1.30.0 branch, but not in the master branch.

knnniggett, 2016-06-11:
zmupdate.pl shows the following non-fatal warning every time it is run:
```
Subroutine loadConfigFromDB redefined at /usr/share/perl5/vendor_perl/ZoneMinder/ConfigAdmin.pm line 69.
Subroutine saveConfigToDB redefined at /usr/share/perl5/vendor_perl/ZoneMinder/ConfigAdmin.pm line 118.
Subroutine main::loadConfigFromDB redefined at /bin/zmupdate.pl line 75.
	main::BEGIN() called at /bin/zmupdate.pl line 75
	eval {...} called at /bin/zmupdate.pl line 75
Subroutine main::saveConfigToDB redefined at /bin/zmupdate.pl line 75.
	main::BEGIN() called at /bin/zmupdate.pl line 75
	eval {...} called at /bin/zmupdate.pl line 75
```

SteveGilvarry, 2016-06-12:
@connortechnology can we do something to solve the ZMS_PATH change that everyone has to make? Assuming /zm/cgi-bin is the final resting place then we should make ZMS_PATH match apache conf.

pliablepixels, 2016-06-12:
I'm almost tempted to say we should just configure the PATH_ZMS based on ``lsb_release -d`` post install - since 99% of the folks don't tinker with install paths

connortechnology, 2016-06-12:
Hmm ... yes. Zms. WE should be able to do something. Like script a test and automatic update.

abishai, 2016-06-13:
Will it break something? We use WWWROOT/cgi-bin ...

connortechnology, 2016-06-13:
@abishai this would be part of the debian package scripts, and would not affect freebsd.

SteveGilvarry, 2016-06-21:
I have added a few existing PR's to the 1.30 milestone, mainly where they clearly fix existing bugs. Close these out and we can RC2, any other issues or PR's that should be in now?

abishai, 2016-06-21:
What about https://github.com/ZoneMinder/ZoneMinder/issues/1542 ? That's obvious and simple to fix issue. Or it's not confirmed?

josh4trunks, 2016-06-23:
It would be nice to get this PR in so admins could specify their mysql port/socket
https://github.com/ZoneMinder/ZoneMinder/pull/1498

I understand we want to consolidate the database connections for Perl after 1.30, but merging this fix doesn't make that more difficult. It just makes the database connection logic identical and correct in the 3 Perl files currently making the connections.

knnniggett, 2016-06-24:
Everyone of us can sit and make a never ending list of things each of us wants to include in the next release. I can do it too. This is exactly why we haven't actually released 1.30.0 yet. It literally never ends. As soon as we merge one guy's pet project, someone else shows up and makes the same argument to merge his or her pet widget. We have to draw the line somewhere if we are ever going to call 1.30.0 a release. 

So the answer to "what about...." is no. Wait until the next release after 1.30.0. Yes, I'm being a hard ass about this, and I'm not going to enter into a discussion about it because my time it very valuable to me.

josh4trunks, 2016-06-24:
I can wait, I'll continue to apply the patches to my own production system. Figured I'd present it since Steve asked.

No worries, I'd prefer ZM release often and not be like those projects which don't release an official stable version for years because they lost their cadence.

onlyjob, 2016-06-26:
:+1: @knnniggett. I'd rather see more point releases with small changes rather than infrequent releases with all the stuff that everybody wants in...
Could it be that some of us have this "last chance" feeling because releases don't happen often enough?

knnniggett, 2016-07-04:
Discovered a show stopper for all systems running the latest version of libjpeg-turbo 1.5.0.
The latest versions of Fedora, Arch Linux, and possibly Debian (@onlyjob can you check ....libjpeg8?) already ship with this version of libjpeg-turbo so it is imperative we get this into master and test it. 
See #1548 

When we run into an issue like this, not only do we need to merge it, but we need to allow sufficient time for it to be tested by the community. This unfortunately means it will be a bit longer before 1.30.0 is released.

I do intend to create an RC2 once #1548 is merged.

onlyjob, 2016-07-05:
Debian moved to _libjpeg-turbo_ v1.5.0 about a month ago: https://tracker.debian.org/pkg/libjpeg-turbo
_libjpeg8_ is deprecated not used by default in Debian. _libjpeg8_ is not the same as _libjpeg-turbo_...

SteveGilvarry, 2016-07-06:
@onlyjob tried to create alioth account to submit this but no joy tonight. New dependencies from zmtelemetry.pl.
libsys-cpu-perl
libsys-meminfo-perl

SteveGilvarry, 2016-07-06:
@knnniggett If possible can we document so others can follow suit. And check out annotated tags so git describe works for release with having to add --tags.
Currently we get:
git describe
v1.27.0-2632-g75a9860
git describe --tags
v1.30.0-rc1-37-g75a9860
So last annotated tag was 1.27.0, 2632 commits ago.
https://nitstorm.github.io/blog/introduction-to-git-tags/

knnniggett, 2016-07-06:
@SteveGilvarry I will add "update the release documentation in the wiki" to my list of things I can do when I'm not playing fireman. I do appreciate your interest and look forward to having help. However, I don't expect to get to that until after the 1.30.0 release. The good news is there are far fewer steps than what used to be. 

Due to time constraints, I'm going to create the next RC2 tag via the normal method. 

I see what you are saying about annotated tags, but I could use some help understanding exactly what this seemingly redundant release information is useful for. I'm not saying I won't do it, but knowing some use cases will help me figure out what information to put in it.

Here is the most recent annotated tag from v.127:
```
tag v1.27.0
Tagger: Kyle Johnson <kjohnson@gnulnx.net>
Date:   Sat Mar 15 00:51:19 2014 -0400

Release 1.27.0
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1.4.11 (GNU/Linux)

iQIcBAABAgAGBQJTI9xHAAoJEOARQ6oZybwBnB8P/1zul6IdWuIGDoJrz2dNsnjA
mB6JNHXktbNfHGuAt9nBPnBfF6gkzKJcUHTk7aaIjm345rK6XgjM848ikZNLxEPG
H4Q0hZFaXG9N5/EiD1PW2OTjveaPzwBK0V/lSQQG/FQrI4i8A2YO/pHRYAP/xdJC
3FuXVLz1Y/ppBSIrxVATKiHirEKoktsfw9hixxne1XfvAlcF9K5pfolOYTzfQ0oj
7Rve6NGhWNw2GyBqovFp3HmjSRV8/hHgTPB4jXMAgTBiko4jnZrAvy/756ooo1Eb
wvqS/bagyuxlpZ0w6XXxA+g/Jm/k0VbdFTSM6iYq+LHWV5+LlPFbB4QwHQIWYWqU
m2V8AR7EQjpWUykHiXr/C8GbkijYbIEqDtfZWDNiftuyfk/ZJAiMpbeTMzHOgmtr
JbqKkFofpccKU5hbmaTPd31Ld4AUk96W6a6J9r/6MPhMZYtAczv72fjtoPaioOse
Ckqe3iYy2SWtUS0BmOFPcr9x0hm9dqspIPuX8Ei2H+P5zCZoIZXkDJRWYr/ycsnY
mHyll8YPb0l9OsruAJPNUiNbzii6HdypWN96IxtA0Gq2sP8mmlUqeF1HObUdQDW+
fWsLLwA0XOEWqil45j6xoEn9be7gWS/weZwhSBW/9PHp9LLr6k6pn4t8Fki8HfLV
CwnnQ1GoQcMOqomUBcCo
=Yse8
-----END PGP SIGNATURE-----

commit 75d390d70aff164f9afdeead6f5a344c8063e8ae
Merge: 16c59f7 d37b4a2
Author: Kyle Johnson <kjohnson@gnulnx.net>
Date:   Sat Mar 15 00:51:13 2014 -0400

    Merge branch 'release-1.27'
```
I don't find this particular information to be too useful, do you? So let's decide what to put in the annotation, and I'll do it for the 1.30.0 release.

SteveGilvarry, 2016-07-06:
Let's me use git describe for versioning, and gives count of commits since
last version tag. I found it when using an example to automate packaging,
and I was scratching head why it kept wanting to generate 1.27. Not sure of
use beyond that but seems annotated is more proper for releases over
lightweight tags.

knnniggett, 2016-07-06:
Okay, so RC2 is out. 
Based on recent issues, it looks like we need to look at exports again (leaves fireman hat on).

I made the mistake of leaving my .git folder in the tarball attached to draft release.... note to self. Don't do that. It ballooned the archive by a factor of 10.

@SteveGilvarry That sounds reasonable, but what do you want the contents of the annotation to be? 
Would this be sufficient?
```
tag v1.27.0
Tagger: Kyle Johnson <kjohnson@gnulnx.net>
Date:   Sat Mar 15 00:51:19 2014 -0400
```
And can one add an annotation after the tag has already been created?

SteveGilvarry, 2016-07-06:
Yes same as the 1.27 one works fine. I think I saw something about
replacing a tag.

knnniggett, 2016-07-06:
Force updated the tag like so:
http://stackoverflow.com/questions/5002555/can-a-lightweight-tag-be-converted-to-an-annotated-tag

It looks like it worked, but it is ugly:

```bash
abauer@Ubuntu-Desktop:~/git/ZoneMinder$ git describe
v1.30.0-rc2
abauer@Ubuntu-Desktop:~/git/ZoneMinder$ git show v1.30.0-rc2
tag v1.30.0-rc2
Tagger: Andy Bauer <knnniggett@hotmail.com>
Date:   Wed Jul 6 18:53:33 2016 -0500

tag v1.30.0-rc2
Tagger: Andrew Bauer <knnniggett@users.sourceforge.net>
Date:   Wed Jul  6 18:49:07 CDT 2016

tag v1.30.0-rc2
Tagger: Andy Bauer <knnniggett@hotmail.com>
Date:   Wed Jul 6 18:48:37 2016 -0500

tag v1.30.0-rc2
Tagger: Andrew Bauer <knnniggett@users.sourceforge.net>
Date:   Wed Jul  6 18:49:07 CDT 2016

commit 4873683f01eb66634dd857ef7904f5c9aaeb01dc
Merge: 257ce2b 75a9860
Author: Steve Gilvarry <SteveGilvarry@users.noreply.github.com>
Date:   Thu Jul 7 00:13:54 2016 +1000

    Merge pull request #1548 from knnniggett/libjpegturbo
    
    fix jpeg buffer too small
```

SteveGilvarry, 2016-07-07:
OK so hopefully a normal annotated tag does it cleaner, but desired result is achieved, so thanks for giving it a go.

And were you referring to the export email thread in the forum when you said issues?

knnniggett, 2016-07-07:
#1550 
The same code that handles exports could be called to create the email attachment.... at least that is my guess what to look for if/when I've got a moment to look into it.

onlyjob, 2016-07-07:
> seems annotated is more proper for releases over lightweight tags.

Not sure if this is relevant here but I've seen `dh-make-golang` complaining about lightweight tags:

> Lightweight tags (created by e.g. "git tag") are problematic because, among other
> things, they are ignored by "git describe" by default.
> Please suggest that the upstream replaces the lightweight tags with annotated ones.
> See https://github.com/russross/blackfriday/issues/196 for details.

[2 more replies]

closed 2016-07-28 by knnniggett