ZoneMinder/zoneminder #1493: API - Add Monitor returns Undefined index: Device

Glyphs, 2016-05-19:
Hello,

Whenever I try adding a new monitor through the API, the monitor is added without any information filled in. 
[image]

The webserver responds with a 200 (or 204? I accept both in my wrapper) but when I pull the actual response message out I see the following information:

`Notice (8): Undefined index: Device [APP/Controller/MonitorsController.php, line 243]
Code Context
MonitorsController::daemonControl() - APP/Controller/MonitorsController.php, line 243
MonitorsController::add() - APP/Controller/MonitorsController.php, line 106
ReflectionMethod::invokeArgs() - [internal], line ??
Controller::invokeAction() - CORE/Cake/Controller/Controller.php, line 490
AppController::invokeAction() - APP/Plugin/Crud/Lib/CrudControllerTrait.php, line 31
Dispatcher::_invoke() - CORE/Cake/Routing/Dispatcher.php, line 191
Dispatcher::dispatch() - CORE/Cake/Routing/Dispatcher.php, line 165
[main] - APP/webroot/index.php, line 108`

The data I am POST-ing to monitors.json is as follows:

`Monitor[Name]=Demo-Unit&Monitor[ServerId]=0&Monitor[Type]=Local&Monitor[Function]=Modect&Monitor[Enabled]=0&Monitor[Channel]=0&Monitor[Format]=0&Monitor[V4LMultiBuffer]=False&Monitor[V4LCapturesPerFrame]=0&Monitor[Port]=0&Monitor[Width]=0&Monitor[Height]=0&Monitor[Colours]=0&Monitor[Deinterlacing]=0&Monitor[RTSPDescribe]=False&Monitor[Brightness]=0&Monitor[Contrast]=0&Monitor[Hue]=0&Monitor[Colour]=0&Monitor[LabelX]=0&Monitor[LabelY]=0&Monitor[LabelSize]=0&Monitor[ImageBufferCount]=0&Monitor[WarmupCount]=0&Monitor[PreEventCount]=0&Monitor[PostEventCount]=0&Monitor[StreamReplayBuffer]=0&Monitor[AlarmFrameCount]=0&Monitor[SectionLength]=0&Monitor[FrameSkip]=0&Monitor[MotionFrameSkip]=0&Monitor[AnalysisFPS]=0&Monitor[AnalysisUpdateDelay]=0&Monitor[MaxFPS]=0&Monitor[AlarmMaxFPS]=0&Monitor[FPSReportInterval]=0&Monitor[RefBlendPerc]=0&Monitor[Controllable]=0&Monitor[ControlId]=0&Monitor[AutoStopTimeout]=0&Monitor[TrackMotion]=0&Monitor[TrackDelay]=0&Monitor[ReturnLocation]=0&Monitor[DefaultRate]=0&Monitor[DefaultScale]=0&Monitor[Exif]=False&Monitor[Sequence]=0`

As you can see, this does not include the Device option. Am I doing something wrong, or does this option in the API not work?

pliablepixels, 2016-05-19:
I'm not sure what the exact command is, but there are two issues:

a) This works:
```
curl -k -b cookies.txt https://myserver:myport/zm/api/monitors.json -d"Monitor[Name]=test1-hyphen&Monitor[Function]=Modect&Monitor[Type]=Remote&Monitor[Protocol]=http&Monitor[Method]=simple&Monitor[Host]=a:b@1.1.1.1"
```

Produces:
[image]

b) However, the start to daemonControl is wrong, which is why after adding the monitor, it throws an error. To fix that, edit MonitorController.php, around line 106, remove the ``$this->request->data`` 

``
//$this->daemonControl($this->Monitor->id, 'start', $this->request->data);
$this->daemonControl($this->Monitor->id, 'start');
``

Glyphs, 2016-05-19:
I'll have to try that later, not currently at a location where I have access to a ZM installation.

Anyhow, I'm not using a command from the linux box itself, so no curl involved. I've basically took what the documentation stated, and converted this to a C# method. A simple webrequest which POSTs data to the system with the data as its content. I'll have a fidget with that some more after I fix up MonitorController.php.

closed 2016-05-22 by connortechnology

Glyphs, 2016-05-25:
Just confirming that this did indeed fix the issue.

PR #1494 (2016-05-22): don't pass partial data to daemoncontrol
seems to resolve #1493 - same problem as a previous fix by iconnor