ZoneMinder/zoneminder #1509: SIGBUS on armv6l (Raspberry Pi)

rtsisyk, 2016-06-01:
```
root@pi:~# gdb --args zmc -m 1
(gdb) r
Starting program: /usr/bin/zmc -m 1
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGBUS, Bus error.
0xb6ff23dc in memset () from /usr/lib/arm-linux-gnueabihf/libcofi_rpi.so
(gdb) bt
#0  0xb6ff23dc in memset () from /usr/lib/arm-linux-gnueabihf/libcofi_rpi.so
#1  0x2a027a38 in Monitor::Monitor(int, char const*, unsigned int, int, bool, char const*, Camera*, int, unsigned int, char const*, char const*, Coord const&, int, int, int, int, int, int, int, int, int, int, double, unsigned int, int, int, int, int, int, bool, unsigned int, bool, Monitor::Purpose, int, Zone**) ()
#2  0x2a02c7ac in Monitor::Load(unsigned int, bool, Monitor::Purpose) ()
#3  0x2a015c0c in main ()

```

zone minder was installed from Debian Stretch packages:

```
 apt-cache policy zoneminder
zoneminder:
  Installed: 1.29.0+dfsg-2
  Candidate: 1.29.0+dfsg-2
  Version table:
 *** 1.29.0+dfsg-2 990
        990 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status
```

```
zmc --version
1.29.0
```

```
uname -a
Linux pi 3.6.11+ #371 PREEMPT Thu Feb 7 16:31:35 GMT 2013 armv6l GNU/Linux
```

connortechnology, 2016-06-01:
I suspect you do not have enough ram.  Or enough ram disk space for the shm files in /dev/shm.

Reduce buffers or resolution.

rtsisyk, 2016-06-02:
SHM is already increased. How to reduce buffers?
I have 512M on this device and I want to add 1980x1080 camera. Is it possible?

connortechnology, 2016-06-02:
edit the monitor, go to the buffers tab.  it defaults to 50.  50 buffers at 1920x1080 32 bit is 41472000 bytes.  That's pretty much all of your ram.  Try reducing all buffers to the minimum.

rtsisyk, 2016-06-04:
Thanks, reducing the number of frames to 10 works for me. 
It seems that I need more powerful hardware for processing.

closed 2016-06-04 by rtsisyk