ZoneMinder/zoneminder #1510: mysql57 - header files detection problem

abishai, 2016-06-01:
I have a problem compiling ZM with mysql57 header files:
-- Looking for mysql/mysql.h
-- Looking for mysql/mysql.h - not found

Run Build Command:"/usr/local/bin/gmake" "cmTC_81d08/fast"
gmake[2]: Entering directory '/usr/home/abishai/zoneminder/work/ZoneMinder-1.30.0-rc1/CMakeFiles/CMakeTmp'
/usr/local/bin/gmake -f CMakeFiles/cmTC_81d08.dir/build.make CMakeFiles/cmTC_81d08.dir/build
gmake[3]: Entering directory '/usr/home/abishai/zoneminder/work/ZoneMinder-1.30.0-rc1/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_81d08.dir/CheckIncludeFile.c.o
/usr/bin/cc   -I/usr/local/include  -O2 -pipe  -fstack-protector -fno-strict-aliasing    -o CMakeFiles/cmTC_81d08.dir/CheckIncludeFile.c.o   -c /usr/home/abishai/zoneminder/work/ZoneMinder-1.30.0-rc1/CMakeFiles/CMakeTmp/CheckIncludeFile.c
In file included from /usr/home/abishai/zoneminder/work/ZoneMinder-1.30.0-rc1/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:
In file included from /usr/local/include/mysql/mysql.h:64:
/usr/local/include/mysql/mysql/client_plugin.h:103:10: fatal error: 'mysql/plugin_auth_common.h' file not found
#include <mysql/plugin_auth_common.h>
         ^
1 error generated.

The issue that in mysql57 devs added some includes in mysql.h - and the mysql/plugin_auth_common.h is out of scope with checks ZM implements.

The idea is to use hints in find_path.

        find_path(MYSQLCLIENT_INCLUDE_DIR mysql.h /usr/local/include/mysql /usr/include/mysql)
        if(MYSQLCLIENT_INCLUDE_DIR)
                include_directories("${MYSQLCLIENT_INCLUDE_DIR}")
                set(CMAKE_REQUIRED_INCLUDES "${MYSQLCLIENT_INCLUDE_DIR}")
        endif(MYSQLCLIENT_INCLUDE_DIR)
        mark_as_advanced(FORCE MYSQLCLIENT_LIBRARIES MYSQLCLIENT_INCLUDE_DIR)
        check_include_file("mysql.h" HAVE_MYSQL_H)

I was able to compile ZM after that.

knnniggett, 2016-06-07:
ZoneMinder does indeed build against mysql 5.7 on other distros. Ubuntu 16.04, which seems to be the most popular distro to use with zoneminder, ships by default with mysql 5.7. bbunge has even written documentation on how to do it.

Before we can say whether a change is needed for your case, we have to figure out what is different in your environment, and that is something I do not currently have time for. Someone will get to this in time.

MrDini123, 2016-11-26:
Hi!

I also have similar problem... The cmake Can't found the mysql/mysql.h file. But I have mysql 5.6.14 and it has a mysql.h header file. But it isn't at /usr/include/mysql/mysql.h, it is located at /ffp/include/mysql.h... So, is there a way to solve this error?

Ps: I am using the latest (1.30 source) stable version of ZoneMinder.

Many thanks!

abishai, 2016-11-27:
I patched CMake files to search header files in multiple locations
https://svnweb.freebsd.org/ports/head/multimedia/zoneminder/files/patch-CMakeLists.txt?revision=416415&view=markup

MrDini123, 2016-11-27:
Hi,

Thanks for the answer!

I also did the same, but nothing happened. So I looked on the debug output, and it looked like the cmake also looks for the mysql.h at the wrong path. So I removed the CMakeCache.txt and now it is worked! :)

Thanks again!

knnniggett, 2016-11-27:
> I patched CMake files to search header files in multiple locations
> https://svnweb.freebsd.org/ports/head/multimedia/zoneminder/files/patch-CMakeLists.txt?revision=416415&view=markup

And you didn't tell us? 
Would appreciate a pull request.

> Hi,
> 
> Thanks for the answer!
> 
> I also did the same, but nothing happened. So I looked on the debug output, and it looked like the cmake also looks for the mysql.h at the wrong path. So I removed the CMakeCache.txt and now it is worked! :)
> 
> Thanks again!

When developing a build with CMake, you have to delete the CMakeCache.txt file each and every time before restarting the build. Storing your config in a cache file is by cmake design. It won't change the value until you delete the cache file.

MrDini123, 2016-11-27:
Thanks for the info. Interesting! :)

I am not familiar in cmake, I am just learning that, but I am missing a function, like distclean... But the Cache file remove did the trick, So now everything is okay!

knnniggett, 2016-11-27:
No problem. Many people just starting off with cmake don't realize that. I'm not sure I've seen it documented anywhere, and I thought I'd mention it so you can avoid any future problems.

I'm hoping @abishai submits a pull request for any freebsd patches he has. We always need volunteers willing to help out... and just doing the work of creating the pull request helps us greatly.

abishai, 2016-11-27:
Well, FreeBSD build infrastructure has this patches, so all FreeBSD folks have access to ZM.
The problem is to make patches compatible with Linux, I'm not Linux guy and it's very hard to check if it still builds. But I can try.

MrDini123, 2016-11-27:
I have to manually patch the CMakeList.txt, because mine mysql.h header isn't in include/mysql/ folder, just in include/.

And on my system, everything is at /ffp prefix, So for example my include dir is /ffp/include.

I think, that caused the problem.

knnniggett, 2016-11-27:
Changes to CMakeLists.txt is fairly universal. We should be able to get CMake to work for all unixes.
Any C related patches might be more difficult.

knnniggett, 2017-05-18:
I believe we fixed this in #1761 
If I am mistaken then let me know and we can take another look

closed 2017-05-18 by knnniggett