ZoneMinder/zoneminder #1512: Ongoing Modetect recordings get cancelled on camera disconnect

SharkWipf, 2016-06-02:
If you configure Zoneminder to record on motion, and it loses connection to the camera while triggered to record, it'll stop recording and not save the recorded event.

This means that a malicious person could effectively just unplug/otherwise disable the camera without raising alarms, even while it is already recording them.
Kinda makes the concept of a "security camera" a lot less secure.

Tested with:  
Source Type: Remote
Remote Protocol: HTTP
Remote Method: Simple
Remote Host Path: /video1.mjpg

SharkWipf, 2016-06-03:
After I unplugged my camera yesterday, I left it unplugged. 
After 24 hours and 10 minutes, I got an email from Zoneminder saying it detected motion, linking to a recording of me unplugging the camera.
Sounds to me like it does record the video, crashes on disconnect and later picks up the pieces again.
However, some earlier disconnect recordings did not go through, even after 48+ hours.

0xABAB, 2016-07-08:
ZoneMinder appears to be developed by people without credentials and in general they don't seem to know what they are doing. Expecting ZoneMinder to work is a mistake.

SharkWipf, 2016-07-08:
#salty

closed 2017-03-18 by SteveGilvarry