ZoneMinder/zoneminder #1513: sudo?

dalaimana, 2016-06-02:
Starting ZoneMinder: Can't exec "sudo": No such file or directory at /usr/share/perl5/ZoneMinder/General.pm line 110.

Zoneminder 1.30 from Christian Marillat deb-multimedia.org
Use of uninitialized value $output in scalar chomp at /usr/share/perl5/ZoneMinder/General.pm line 119.
Use of uninitialized value $output in concatenation (.) or string at /usr/share/perl5/ZoneMinder/General.pm line 120.
failure

I never install sudo in a server machine..

josh4trunks, 2016-06-03:
agreed that this seems primitive, anyway 'su' could be used instead? Wold be nice to remove the 'sudo' requirement from the freebsd port as well.

onlyjob, 2016-08-01:
@dalaimana, it is worth mentioning that deb-multimedia.org is a deprecated repository on Debian... Zoneminder is available from official Debian repositories so (to say the least) it is strange to install foreign package...
Also those packages are not maintained here. If you are concerned about DMO packages then perhaps you'd better contact DMO maintainers...

dalaimana, 2016-09-22:
@onlyjob nope, deb-multimedia.org is NOT a deprecated repository.. sorry for you.

onlyjob, 2016-09-22:
deb-multimedia.org is an unofficial (3rd party) repository, unsupported by Debian:

  https://wiki.debian.org/DebianMultimedia/FAQ#There_is_.27Debian_Multimedia_Maintainers.27_and_.27deb-multimedia.org.27._So_what.27s_the_difference.3F

Once can consider all 3rd party repositories to be deprecated on Debian but deb-multimedia.org _is_ deprecated in favour of official packages maintained by multimedia team. Formerly (years ago) deb-multimedia.org was the source of many packages that are not available on Debian. These days most multimedia packages are available from official repositories. People often run into problems when they install packages from 3rd party repositories...

dalaimana, 2016-10-05:
onlyjob, yes yes.. but not is deb-multimedia the problem, but sudo!

onlyjob, 2018-06-16:
No activity doesn't mean that the problem is gone.

closed 2018-08-22 by stale[bot]