ZoneMinder/zoneminder #1516: JPEG file quality unaffected by alarm

candhill, 2016-06-03:
zm v1.29.0 on Ubuntu 14.04 LTS 3.13.0-86-generic using ppa:iconnor/zoneminder using a 1080p (Realtek?) USB camera which gets around 4.3fps using mocord.

When I set jpeg_file_quality at 100 each frame is around 1.2MB.  If I set it to 70 each frame is around 112KB.  No matter what I set jpeg_alarm_file_quality to the jpeg is always the same size as the jpeg_file_quality setting.

1) going from 1.2MB to 112KB is a huge difference for being 70% quality.  Does this sound right?

If someone breaks in I would very much want the absolute best quality possible to catch the bad guy.  With such space savings using 70% without alarm it's hard to pass that up for "worthless" frames (I only save them in case the pre and post alarm frames aren't enough to capture all the action/relevant frames...and I have the space).  

2) So, jpeg_alarm_file_quality does not appear to have any affect and it appears to be a bug.

Please advise.

Thanks,

SteveGilvarry, 2016-06-04:
So something like this is what you are doing? Help for alarm quality indicates it would only work if value is higher than standard setting.
jpeg_file_quality = 70
jpeg_alarm_file_quality = 100

candhill, 2016-06-04:
Yep, exactly.  With those settings all jpeg frames are around 112KB, even if there's an alarm.  If I leave alarm quality 100 and set jpeg quality to 100 all jpeg frames are around 1.2MB.  The alarm quality setting doesn't seem to make any difference.

Thanks

SteveGilvarry, 2016-06-04:
At a glance the code looks ok, will try debugging it tomorrow, once I turn jpegs back on.
https://github.com/ZoneMinder/ZoneMinder/blob/66a3fef61ebd10c45968a3706021d45d027e28be/src/zm_event.cpp#L306

closed 2018-06-22 by stale[bot]