ZoneMinder/zoneminder #1517: do_debian.sh version name for package

pliablepixels, 2016-06-04:
do_debian uses zoneminder_release/version to determine package name version. Given our recent discussion about not bumping up version in rc builds, this leads to a confusing representation where the package that is built is called 1.29 but is actually 1.30 (in the case where 1.30 is not stable).

Given that @connortechnology uses this file for local version changes, does it make sense to add an option to do_debian that says "gitversion" or something that replaces [VERSION](https://github.com/ZoneMinder/ZoneMinder/blob/master/utils/do_debian_package.sh#L40) with the value of ``git describe --tags $(git rev-list --tags --max-count=1)`` ? That way for others building a package from source, the output is always consistent. Thoughts?

knnniggett, 2016-06-04:
This is the same conversation that is discussed in #1515 
See that issue for the proposed solution which will solve this.

pliablepixels, 2016-06-04:
One of the reasons I made this a separate issue was it is not clear to me if those discussions resolve this particular issue. If it is, I'll close this.

My understanding so far is you will not be changing the release version in the version file for pre-releases  (irrespective of whether the version file is in a wiki page or not). Is that correct? If so, this issue will continue?

knnniggett, 2016-06-04:
Yes, this is exactly the problem @connortechnology described in #1485. This is why even he can't build a package right now. The solution is shown in #1515 . 

Let's not unnecessarily spawn more issues surrounding the same topic. You can always ask the question within the existing issue if it applies or not.

pliablepixels, 2016-06-04:
Happy to close this, but you haven't answered my core question - will you be updating the version text for pre-releases as well (like 1.30 right now) ? If so, that PR will solve it. If not, then this is a different issue, not related to the PR you pointed me to.

closed 2016-06-04 by pliablepixels