ZoneMinder/zoneminder #1518: Zone area size not updated (release 1.30)

manciuleas, 2016-06-06:
I have upgraded my ZoneMinder installation from 1.29 to 1.30 (on Slackware 14.2rc1) and I noticed that when creating a monitor zone (smaller than the image size) for a modect monitor the area size is still equal to the total number of pixels in the image (see attached picture).
[image].  
This functionality used to work in ZM 1.29 as I have a monitor with 4 zones created with that version and the area of the zones is reported correctly.
I have tried to debug the issue and found out that in the submitForm( form ) function (views/js/zone.js), the zone.Area variable holds the area of the whole image instead of the area of the defined zone, event though in views/zone.php getPolyArea() is called to computed the actual area of the zone (it seems like the global 'zone' variable is  not updated).

[image]

knnniggett, 2016-06-06:
Confirmed. Since we switched zone drawing to svg, something is not triggering to update the area like it should. I have not found where yet.

manciuleas, 2016-06-07:
I've looked a bit at the source code and I noticed that in ZM 1.29 skins/classic/views/js/zone.js contains a function (updateZoneImageResponse) that updates the zone.Area variable in response to an updateZoneImage() request. If my understanding is correct updateZoneImage() sends the request to the server which uses the PHP code to compute the area and returns it in the AJAX response to the client. In ZM 1.30, however, it looks like the update of the zone image is local (to the client) and thus no one updates the value of zone.Area.
As a temporary workaround I have changed the added the following core in zone.js.php:
zone['Points'] = new Array();
<?php
for ( $i = 0; $i < count($newZone['Points']); $i++ )
{
?>
zone['Points'][<?php echo $i ?>] = { 'x': <?php echo $newZone['Points'][$i]['x'] ?>, 'y': <?php echo $newZone['Points'][$i]['y'] ?> };
<?php
}
>>>>>> $zone['Area']=getPolyArea( $newZone['Points'] ); <<<<<
?>    
>>>>> zone['Area']=<?php echo $zone['Area'] ?>; <<<<
With this code the size of the zone area seems to be computed correctly but only if I edit the zone twice:
1. Edit zone -> save -> area value in the zone window is unchanged
2. Edit zone (don't change anything) -> Save -> area value is updated.

Hope this helps to find the proper fix,

manciuleas, 2016-06-07:
I've looked a bit at the source code and I noticed that in ZM 1.29 skins/classic/views/js/zone.js contains a function (updateZoneImageResponse) that updates the zone.Area variable in response to an updateZoneImage() request. If my understanding is correct updateZoneImage() sends the request to the server which uses the PHP code to compute the area and returns it in the AJAX response to the client. In ZM 1.30, however, it looks like the update of the zone image is local (to the client) and thus no one updates the value of zone.Area.
As a temporary workaround I have changed the added the following core in zone.js.php:

    zone['Points'] = new Array();
    <?php
    for ( $i = 0; $i < count($newZone['Points']); $i++ )
    {
    ?>
    zone['Points'][<?php echo $i ?>] = { 'x': <?php echo $newZone['Points'][$i]['x'] ?>, 'y': <?php echo    $newZone['Points'][$i]['y'] ?> };
    <?php
    }
    >>>> $zone['Area']=getPolyArea( $newZone['Points'] ); <<<<<
    ?>
    >>>> zone['Area']=<?php echo $zone['Area'] ?>; <<<<

With this code the size of the zone area seems to be computed correctly but only if I edit the zone twice:
1. Edit zone -> save -> area value in the zone window is unchanged
2. Edit zone (don't change anything) -> Save -> area value is updated.

knnniggett, 2016-06-07:
see #1521

manciuleas, 2016-06-07:
Thank you. I noticed it too - waiting for the check-in :)
Maybe you can move the division by 2 at the very end to speed up computations a bit,

knnniggett, 2016-07-06:
We've tested this to be working in the latest master so I'm going to close this out.
1.30.0 RC2 is on the way...

closed 2016-07-06 by knnniggett