ZoneMinder/zoneminder #1519: 1.30 fail to build without v4l

abishai, 2016-06-06:
v4l is an optional part of ZM, however https://github.com/ZoneMinder/ZoneMinder/blob/66a3fef61ebd10c45968a3706021d45d027e28be/src/zm_local_camera.h#L94 uses VIDEO_MAX_FRAME which is not available on systems without v4l support

We need to define it manually if v4l is not available.

knnniggett, 2016-06-06:
While you can build zoneminder without v4l, it is by design that local devices will not work without v4l:
https://github.com/ZoneMinder/ZoneMinder/blob/master/CMakeLists.txt#L551

abishai, 2016-06-06:
I understand now. So strange, FreeBSD build infrastructure failed to build ZM port for i386 and I can't recreate it on my build environment yet.
http://package19.nyi.freebsd.org/data/103i386-default-build-as-user/416439/logs/zoneminder-1.30.0.log

abishai, 2016-06-06:
Ah, I see
-- Looking for libv4l1-videodev.h
-- Looking for libv4l1-videodev.h - found
-- Looking for linux/videodev2.h
-- Looking for linux/videodev2.h - not found

I wonder where it found libv4l1-videodev.h

Mine build shows 

-- Looking for libv4l1-videodev.h
-- Looking for libv4l1-videodev.h - not found
-- Looking for linux/videodev.h
-- Looking for linux/videodev.h - not found
-- Looking for linux/videodev2.h
-- Looking for linux/videodev2.h - not found

abishai, 2016-06-06:
I understand now - with such options, zm_local_camera is included, however VIDEO_MAX_FRAME lives in videodev2.h 
Nasty!

knnniggett, 2016-06-06:
Perhaps v4l support got pulled in due to a package dependency. I see this earlier in your build output:
```
===>   zoneminder-1.30.0 depends on executable: ffmpeg - not found
===>   Installing existing package /packages/All/ffmpeg-2.8.7_1,1.txz
[103i386-default-build-as-user-job-16] Installing ffmpeg-2.8.7_1,1...
[103i386-default-build-as-user-job-16] `-- Installing libv4l-1.6.3_1...
[103i386-default-build-as-user-job-16] |   `-- Installing jpeg-turbo-1.4.2...
[103i386-default-build-as-user-job-16] |   `-- Extracting jpeg-turbo-1.4.2: .......... done
[103i386-default-build-as-user-job-16] `-- Extracting libv4l-1.6.3_1: .......... done
[103i386-default-build-as-user-job-16] `-- Installing freetype2-2.6.3...
```

knnniggett, 2016-06-06:
Oh, so VIDEO_MAX_FRAME macro is part of videodev2.h.
Will have to take a look at the logic.

knnniggett, 2016-06-06:
VIDEO_MAX_FRAME appears to be commented out if v4l2 is not detected:
https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_local_camera.cpp#L259

abishai, 2016-06-06:
It ends here https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_local_camera.cpp#L265
and we have .h with this MACRO

abishai, 2016-06-06:
offtopic: If I found you, can you take a look my issue for mysql57 - https://github.com/ZoneMinder/ZoneMinder/issues/1510 I made a local patch for port, however I think it's worth to be upstreamed (if you don't have better solution). https://svnweb.freebsd.org/ports/head/multimedia/zoneminder/files/patch-CMakeLists.txt?revision=416415&view=markup Should I make a pull request or patch?

abishai, 2016-06-06:
Recreated the issue locally. videodev2 is pulled as a build dependency for libv4l1. FreeBSD bot is lazy and use packages, so, obviously doesn't have build deps as it pulls only run-deps.

knnniggett, 2016-06-06:
See #1520

As I wrote that I noticed you may also run into problems with VIDEO_MAX_PLANES as it looks like it also got moved from v4l1 to v4l2.

knnniggett, 2016-06-06:
While we are testing release candidates, we are not merging anything that is not a bug fix.

abishai, 2016-06-06:
But that would be the bug fix :) ZM is not building against mysql 5.7.

knnniggett, 2016-07-04:
Fixed by #1520, closing.

closed 2016-07-04 by knnniggett