ZoneMinder/zoneminder #1530: 1.30 is not compiling under FreeBSD11-ALPHA3

abishai, 2016-06-13:
FreeBSD11 is not yet released. however packages are builded against it as well. 
We need a hint what abs() to use. Can we assume that *psrc is long ?

/wrkdirs/usr/ports/multimedia/zoneminder/work/ZoneMinder-1.30.0-rc1/src/zm_image.cpp:1624:24: error: call to 'abs' is ambiguous
        if ( (unsigned)abs((*psrc)-RGB_VAL(ref_colour,c)) >= RGB_VAL(threshold,c) )
                       ^~~
/usr/include/stdlib.h:83:6: note: candidate function
int      abs(int) __pure2;
         ^
/usr/include/c++/v1/stdlib.h:115:44: note: candidate function
inline _LIBCPP_INLINE_VISIBILITY long      abs(     long __x) _NOEXCEPT {return  labs(__x);}
                                           ^
/usr/include/c++/v1/stdlib.h:117:44: note: candidate function
inline _LIBCPP_INLINE_VISIBILITY long long abs(long long __x) _NOEXCEPT {return llabs(__x);}
                                           ^
/usr/include/c++/v1/math.h:646:1: note: candidate function
abs(float __lcpp_x) _NOEXCEPT {return fabsf(__lcpp_x);}
^
/usr/include/c++/v1/math.h:650:1: note: candidate function
abs(double __lcpp_x) _NOEXCEPT {return fabs(__lcpp_x);}
^
/usr/include/c++/v1/math.h:654:1: note: candidate function
abs(long double __lcpp_x) _NOEXCEPT {return fabsl(__lcpp_x);}

abishai, 2016-06-13:
Or use fabs(), it's less overloaded.

abishai, 2016-06-13:
Or labs() :)
I don't know what solution is better.

abishai, 2016-06-13:
As we already has workaround for SOLARIS here, maybe remove abs entirely if it's so confuse compiler?
        unsigned int diff = ((*psrc)-RGB_VAL(ref_colour,c)) > 0 ? (*psrc)-RGB_VAL(ref_colour,c) : RGB_VAL(ref_colour,c) - (*psrc);
        if (diff >= RGB_VAL(threshold,c))
        {
          count++;
        }

If you don't have better ideas, I can make a PR.

connortechnology, 2016-06-13:
I'm not sure why it is confused, but we don't need the float version of abs.  The values being compared at uint8_t...

abishai, 2016-06-13:
Yeh, I looked into the code lately and rewrited it to dump compare instead of abs() at all.

abishai, 2016-06-13:
Also, the error is gone of I convert *psrc to long.

kylejohnson, 2016-09-02:
@abishai  Did you ever get this resolved?

abishai, 2016-09-02:
I'm patching sources before build.
See https://svnweb.freebsd.org/ports/head/multimedia/zoneminder/files/patch-src_zm__image.cpp?revision=417142&view=markup

However, I'd like it to be upstreamed if you find solution suitable.

closed 2017-05-10 by knnniggett