ZoneMinder/zoneminder #1532: Recompiled FFMPEG like 20 times and still cant install H264 Zoneminder

moeiscool, 2016-06-14:
I have tried to reinsall/build FFMPEG from different places and it still gives me this error when i try to build the h264-feature-storage branch of Zoneminder. Is there a DEB i can install instead? This is getting crazy. The old branch Zoneminder-video based on 1.28 was a lot easier to install. It told me with libs to apt-get when i did check-builddeps or something :( now this one doesn't. Please help this noob!

Steve is probably the guy i should bug but I couldn't find a way to PM him on github.

`[ 84%] Built target zm
Linking CXX executable zma
libzm.a(zm_mpeg.cpp.o): In function `VideoStream::SetupCodec(int, int, int, int, int, double)':
zm_mpeg.cpp:(.text+0xfc3): undefined reference to `avcodec_get_name'
zm_mpeg.cpp:(.text+0x1855): undefined reference to `avcodec_get_name'
zm_mpeg.cpp:(.text+0x1906): undefined reference to `avcodec_get_name'
libzm.a(zm_mpeg.cpp.o): In function `VideoStream::OpenStream()':
zm_mpeg.cpp:(.text+0x50dd): undefined reference to `av_frame_alloc'
zm_mpeg.cpp:(.text+0x514c): undefined reference to `av_frame_alloc'
zm_mpeg.cpp:(.text+0x5af5): undefined reference to `av_frame_free'
zm_mpeg.cpp:(.text+0x5df5): undefined reference to `av_frame_free'
libzm.a(zm_mpeg.cpp.o): In function `VideoStream::~VideoStream()':
zm_mpeg.cpp:(.text+0x85b2): undefined reference to `av_frame_free'
zm_mpeg.cpp:(.text+0x85cc): undefined reference to `av_frame_free'
libzm.a(zm_mpeg.cpp.o): In function `VideoStream::ActuallyEncodeFrame(unsigned char const*, int, bool, unsigned int)':
zm_mpeg.cpp:(.text+0xb7dd): undefined reference to `avcodec_encode_video2'
libzm.a(zm_mpeg.cpp.o): In function `VideoStream::SendPacket(AVPacket*)':
zm_mpeg.cpp:(.text+0xc383): undefined reference to `av_packet_unref'
libzm.a(zm_mpeg.cpp.o): In function `VideoStream::StreamingThreadCallback(void*)':
zm_mpeg.cpp:(.text+0xcb23): undefined reference to `av_packet_unref'
libzm.a(zm_remote_camera_rtsp.cpp.o): In function `RemoteCameraRtsp::~RemoteCameraRtsp()':
zm_remote_camera_rtsp.cpp:(.text+0x117): undefined reference to `av_frame_free'
zm_remote_camera_rtsp.cpp:(.text+0x123): undefined reference to `av_frame_free'
libzm.a(zm_remote_camera_rtsp.cpp.o): In function `RemoteCameraRtsp::CaptureAndRecord(Image&, bool, char*)':
zm_remote_camera_rtsp.cpp:(.text+0xcd9): undefined reference to `av_packet_unref'
zm_remote_camera_rtsp.cpp:(.text+0x1a3e): undefined reference to `av_packet_unref'
libzm.a(zm_remote_camera_rtsp.cpp.o): In function `RemoteCameraRtsp::Capture(Image&)':
zm_remote_camera_rtsp.cpp:(.text+0x3baf): undefined reference to `av_packet_unref'
libzm.a(zm_remote_camera_rtsp.cpp.o): In function `RemoteCameraRtsp::PrimeCapture()':
zm_remote_camera_rtsp.cpp:(.text+0x6ce6): undefined reference to `av_frame_alloc'
zm_remote_camera_rtsp.cpp:(.text+0x6cf2): undefined reference to `av_frame_alloc'
libzm.a(zm_videostore.cpp.o): In function `VideoStore::writeAudioFramePacket(AVPacket*, AVStream*)':
zm_videostore.cpp:(.text+0x7678): undefined reference to `av_get_audio_frame_duration'
libzm.a(zm_ffmpeg.cpp.o): In function `SWScale::SWScale()':
zm_ffmpeg.cpp:(.text+0x2493): undefined reference to `av_frame_alloc'
zm_ffmpeg.cpp:(.text+0x24a5): undefined reference to `av_frame_alloc'
libzm.a(zm_ffmpeg.cpp.o): In function `SWScale::~SWScale()':
zm_ffmpeg.cpp:(.text+0x3600): undefined reference to `av_frame_free'
zm_ffmpeg.cpp:(.text+0x3609): undefined reference to `av_frame_free'
libzm.a(zm_ffmpeg.cpp.o): In function `av_rescale_delta':
zm_ffmpeg.cpp:(.text+0x8615): undefined reference to `av_rescale_q_rnd'
zm_ffmpeg.cpp:(.text+0x863d): undefined reference to `av_rescale_q_rnd'
libzm.a(zm_ffmpeg.cpp.o): In function `zm_dump_stream_format(AVFormatContext*, int, int, int)':
zm_ffmpeg.cpp:(.text+0x892f): undefined reference to `av_opt_ptr'
collect2: ld returned 1 exit status
make[2]: *** [src/zma] Error 1
make[1]: *** [src/CMakeFiles/zma.dir/all] Error 2
make: *** [all] Error 2
`

connortechnology, 2016-06-14:
If you want to live on the bleeding edge, feel free to give my storageareas ppa a try.  ppa:iconnor/zoneminder-storageareas.

It includes the h264 video stuff, plus a ton more stuff that I work on .  It is bleeding edge, but I do use it in production.

moeiscool, 2016-06-14:
thank you kind sir!

moeiscool, 2016-06-14:
in the zm_create.sql
i think github messed up the file on a sync

not sure what to change this part to 

<<<<<<< HEAD
  `Concurrent` tinyint(1) unsigned NOT NULL default '0',
  PRIMARY KEY (`Name`)
=======
  PRIMARY KEY (`Id`),
  KEY `Name` (`Name`)
>>>>>>> filter_id

it keeps failing here on the sql query

moeiscool, 2016-06-14:
events were not saving so i had to manually add "StorageId" as INT (length = 10) column to "Events" Table

connortechnology, 2016-06-15:
welcome to the blleding edge.  I have fixed the merge issue... generating new packages now.

abishai, 2016-06-16:
I can test h264 under FreeBSD if you wish :)

closed 2016-06-16 by SteveGilvarry