ZoneMinder/zoneminder #1535: Update the Installation instructions

deleteZombie, 2016-06-15:
I just had a go at installing this today, it's taken me a few hours to sort through, and I've highlighted some differences which might make life easier for other newcomers below.

The installation instructions for 15.04+ could be updated from:

> if you don't have mysql already installed:
> sudo apt-get install mysql-server

to Install LAMP

> sudo tasksel install lamp-server
> apt-get install php libapache2-mod-php php-mcrypt php-mysql

then the text below can be removed:
> We also need to install php5-gd (as of 1.28.107, this is not installed)
> sudo apt-get install php5-gd

Then update the text:

> Step 5: Edit Timezone in PHP
> vi /etc/php5/apache2/php.ini

to something like:
> find the directory your new php.ini file is in:
> it should be something like /etc / php / [ver] / apache2 / php.ini
> then edit this file
eg:
> vi /etc/php/7.0/apache2/php.ini

Hope this helps

SteveGilvarry, 2016-06-16:
Note to copy current wiki guides https://wiki.zoneminder.com/Contents to installation guide.

SteveGilvarry, 2016-07-05:
Updated Ubuntu and Debian install guides and tested Jessie, 14.04 and 16.04

closed 2016-07-05 by SteveGilvarry