ZoneMinder/zoneminder #1536: ONVIF detection PTZ support

blackPantherOS, 2016-06-17:
Hi

I tried setup a onvif camera (KKMoon 804, PTZ/P2P/RTSP), the ZM detected on: HDIPC @ 192.168.0.102 and the next page after entered cam admin login:

"Unable to probe network cameras, status is '255'"

In log:
"2016-06-17 09:08:25.448899	web_php		19142	FAT	Unable to probe network cameras, status is '255'	/usr/share/zoneminder/www/skins/classic/views/onvifprobe.php	36"

I tried set up other control module for this cam but does not working
ZM version 1.30.0+git

knnniggett, 2016-06-17:
Onvif detection and Onvif PTZ support are two completely separate things.

The only PTZ control script we currently have, which sends ONVIF SOAP commands, is this one:
https://github.com/ZoneMinder/ZoneMinder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/Control/Netcat.pm

This script is bare minimum in terms of functionality, and can certainly use some fine tuning.
Since onvif is a standard, the script stands a good chance to work with other cameras. It just needs to be tested.

As to your other question, you need to run the probe from the command line to get any useful troubleshooting information from it:
```bash
sudo zmonvif-probe.pl probe
```
A common issue is missing perl modules. If that is the error you see, then install the missing modules. If you see something different, then review the existing open onvif issues and find an existing issue to post your data to (just click the ONVIF label on the top right of your screen). 

I'd like to keep duplicate onvif issues to a minimum so I'm going to close this out. If your error really is unique, then feel free to create a new issue once you have data to post.

closed 2016-06-17 by knnniggett

B4unty, 2016-09-11:
Hello I would like to control my Supracam IPC - 100AC over PTZ how should I set it?

Transalpler, 2016-10-30:
I am having the same Problem with ZM 1.30 on Ubuntu LTS 16.04 with a Doorbird Intercom:

`wolfgang@Zoneminder:~$ sudo zmonvif-probe.pl probe
[sudo] Passwort für wolfgang:
http://192.168.27.80/onvif/device_service, 1.1, (Profile='Streaming', hardware='IntercomDoorBirdOne', name='DoorBird', location/country='Germany', location/city='Berlin', location/building='Joachimsthaler_Str_12')
`