ZoneMinder/zoneminder #1537: Zones.json doesn't support pagination

earlymeadow, 2016-06-17:
If I curl http://server/zm/api/zones.json it appears to return a list of all available zones. Now that I've been adding and deleting zones, I noticed that only 20 zones are returned, whilst I have a few more. 

I don't see any pagination item, so there's no way to tell if there are more zones. Also, http://server/zm/api/zones.json?page=1 does nothing more that return the same 20 zones.

I am able to get the neccessary zone info by getting http://server/zm/api/zones/44.json, but seeing as deleting zones leaves holes in the zone numbering (zid), I have no way of knowing when to stop "getting zones" one by one. 

I know I'm using an undocumented API feature, but it would save me a lot of unneccessary calls if it would work as suspected ;).

Am I doing something horribly wrong, or is this a bug?

(using 1.29 cent os 7) (also posted in forum https://forums.zoneminder.com/viewtopic.php?f=34&t=24953)

pliablepixels, 2016-07-14:
The short answer is its likely a bug. I am fixing/extending APIs using zmNinja as the client. So features that zmNinja don't support mostly don't get tested by me. I'll look at the zones API in a few weeks - chances are everything you are saying is correct and there are bugs.

pliablepixels, 2016-09-03:
I've just started looking at zones APIs. If it helps,  to get all the zones for a specific monitor, use `https://server/zm/api/zones/forMonitor/X.json`

PR #1614 (2016-09-07): 1537 zone controller bug
Maybe Fixes #1537 by returning all zones. Doesn't add pagination. Not sure its really needed. 
 I tried with 22 zones and it seems to work.
@earlymeadow - give this a try?
closed 2016-09-07 by connortechnology