ZoneMinder/zoneminder #1542: zmvideo.pl should return filename

abishai, 2016-06-18:
Looks like it's expect that zmvideo.pl returns something, however it's not, so we'll get Video Generation Failed
https://github.com/ZoneMinder/ZoneMinder/blob/master/web/includes/functions.php#L1075
https://github.com/ZoneMinder/ZoneMinder/blob/fe46785e94081d86b6dfb565c423198b52855c77/web/ajax/event.php#L23

knnniggett, 2016-06-18:
You need to tell us exactly what you are trying to do. Did you configure ffmpeg properly under options? That is the most common cause to failed video generation.

abishai, 2016-06-18:
Video generated, however ZM tolds me Video Generation Failed. createVideo() expects that zmvideo.pl will write something in console on success, however it's not (see disabled prints in the end)

0xABAB, 2016-07-10:
`Video generation failed` message also happens with the lowest speed and lowest quality settings when creating an AVI file with correctly configured ffmpeg with 1.29. It does work for a higher quality setting with more frames per second.  

This feels extremely flakey to me. It's like if you look at ZoneMinder wrong, it breaks. Is the original author to blame for this? Is it the ffmpeg people?  

The reported error is way too vague to be useful. While there might be logs, it would be *much* better to just show that in the UI too (e.g. a "details" link).

abishai, 2016-07-10:
ZM uses https://github.com/ZoneMinder/ZoneMinder/blob/master/web/includes/functions.php#L1074 exec() to ezecute external script. It returns the last row from the external script output if execution success. The problem is that zmvideo.pl is not always return output. Actually, in most cases it won't return output to console, so at the end, the file will be created, but Generation Failed error printed.

abishai, 2016-07-10:
@0xABAB you can easily trace the root of the problem by inserting the debug code https://github.com/ZoneMinder/ZoneMinder/blob/master/web/includes/functions.php#L1074 here to grab $command . After that, call the command from the console. It would be interesting to see what it's writing with different settings.

knnniggett, 2016-07-12:
Fixed. See #1557

closed 2016-07-12 by knnniggett