ZoneMinder/zoneminder #1543: (1.30.3 storage areas) Unable to view stream when OPT_USE_AUTH enabled

moeiscool, 2016-06-18:
when i try to view the stream the log shows 
2016-06-18 11:07:36.914646	zms		3712	ERR	Unable to authenticate user	zms.cpp	225
2016-06-18 11:07:36.672888	zms		3711	ERR	Unable to authenticate user	zms.cpp	225
2016-06-18 11:07:36.494288	zms		3710	ERR	Unable to authenticate user	zms.cpp	225
2016-06-18 11:07:36.319226	zms		3709	ERR	Unable to authenticate user	zms.cpp	225
2016-06-18 11:07:36.087327	web_php		3531	WAR	Unknown function call Monitor->CanMoveCon	/usr/share/zoneminder/www/includes/Monitor.php	57
2016-06-18 11:07:36.085793	web_php		3531	WAR	Unknown function call Monitor->CanMoveRel	/usr/share/zoneminder/www/includes/Monitor.php	57
2016-06-18 11:07:36.082661	web_php		3531	WAR	Unknown function call Monitor->CanMoveMap	/usr/share/zoneminder/www/includes/Monitor.php	57

but when i turn off auth and view it works fine. I could not find zms.cpp in the /usr/bin folder... is it somewhere else?

moeiscool, 2016-06-18:
so ive found that if open up the console on the stream
then add the user and pass fields to the url it works.. 
`/zm/cgi-bin/nph-zms?mode=jpeg&scale=100&maxfps=5&buffer=1000&monitor=2&auth=c4bdc1c108f0213469fc97521c1d2110&connkey=515277&rand=1466280768**&user=admin&pass=admin**`

connortechnology, 2016-06-19:
interesting.  I was recently messing around with that code, so I might have broken something.  

Thanks for the report.  I will get to the bottom of it, but likely not till Monday.

connortechnology, 2016-06-20:
zms is located in /usr/lib/zoneminder/cgi-bin

The CanMoveCon etc warnings should be gone in the latest build. 

I can't imagine why auth is not working for you.  It works great for me of course. Did it work for you in master branch?

SteveGilvarry, 2016-07-27:
Same as #1552 going to close this out in favour of the one occurring in master multi-server, but does appear that @connortechnology not seeing this in his installations.

closed 2016-07-27 by SteveGilvarry