ZoneMinder/zoneminder #1551: IP cam stops to be received at random times

energywave, 2016-07-07:
Raspberry Pi2 with 3 720p cam receiving via http picture snapshots and one picam configured at 720p. 1280x720x24bppx20 buffer frames = 52.73MB per cam x 4 = 210.94MB total. I Configured shmmax at 300M and shmall at 73242 (4K pages, for a total of 300M).
I'm receiving at 2 frames x sec (the max is 3) to lower cpu consumption.
All is working great except that at some point a random camera (more frequently the first, but not ever. Happens with modect as well as monitor items) stops to be received. The only way to restart the cam is to restart zoneminder or to save the cam config. (ip camera are still working when this happens, they can be accessed without problems).
I cannot find what the problem is... I've enabled level 5 debug and I waited until that happened but I don't know how to interpret the files and in the forum noone was helping on that: https://forums.zoneminder.com/viewtopic.php?f=34&t=24981
I'm posting here because that could be a zoneminder bug. If it isn't the bug is the lack of information on why that is happening, so something has to change to not drive crazy people :)
P.S.: I was using the same box with Debian stretch and zoneminder 1.28 for one year without an issue.

That said... thank you for your work, zoneminder is great :+1:

energywave, 2016-07-07:
Digging in the logfiles I've found this fragment that seems to be when the camera stopped to be received:
06/29/16 23:40:19.112759 zmc_m4[29959].DB3-zm_remote_camera_http.cpp/147 [Request sent]
06/29/16 23:40:21.615482 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:24.118339 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:26.621128 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:29.123936 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:31.626701 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:34.128694 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:36.631466 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:39.134238 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:41.636151 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:44.138983 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:46.641745 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:49.144519 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:51.646145 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:54.149000 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:56.651775 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:40:59.154594 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:01.657397 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:04.160240 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:06.663009 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:09.164173 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:11.664983 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:14.165844 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:16.666633 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:19.167303 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:21.668078 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:24.168865 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:26.671530 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:29.174352 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:31.677143 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:34.179972 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:36.680807 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:39.183566 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:41.686331 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:44.189161 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:46.691987 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:49.194693 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:51.697420 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:54.200189 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:56.703012 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:41:59.205875 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:42:01.708764 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:42:04.211619 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:42:06.714476 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
06/29/16 23:42:09.216187 zmc_m4[29959].DB4-zm_remote_camera_http.cpp/168 [Select timed out timeout was 0 secs 0 usecs]
[... cut ...]
06/30/16 01:40:28.664074 zmc_m4[29959].ERR-zm_remote_camera_http.cpp/628 [Unable to read header]
06/30/16 01:40:28.681099 zmc_m4[29959].ERR-zm_remote_camera_http.cpp/1141 [Unable to get response, disconnecting]
06/30/16 01:40:28.693769 zmc_m4[29959].DB3-zm_remote_camera_http.cpp/132 [Disconnected from host]
06/30/16 01:40:28.706668 zmc_m4[29959].DB3-zm_remote_camera_http.cpp/132 [Disconnected from host]
06/30/16 01:40:28.814637 zmc_m4[29959].DB1-zm_logger.cpp/239 [Terminating Logger]

I hope this put some light on the issue. Thank you in advance!

knnniggett, 2016-07-07:
What you have described is not a bug with the ZoneMinder source code. The "random camera stopped" issue has got to do with a hardware resource or network bottleneck. It could also be your configuration causing the bottleneck. It is more common on smaller hardware like raspberry pi's because the hardware itself is rather limited.

I'm sorry you have not yet receive help for this issue in the user forum, but double posting into our github forum is not the answer.

closed 2016-07-07 by knnniggett

energywave, 2016-07-07:
My intention wasn't to cross-post the assistance request, I was trying to be constructive, trying to give you some information to improve zoneminder (as I like it). Stated that you support out of the box Raspberry Pi and I followed your wiki guide to set it up and working and, finally, it isn't working stable (as it did 1.28) there should be a problem somewhere...
The problem could be in the wiki doc or in the source, because zoneminder isn't working well. If the problem is bound to low resources then, as I said before, there is a problem in how you write about this problem in the log as it's impossible to understand what is going on (and so to fix it).
So, if you're interested in fixing what is not working on raspberry pi2 I could help by experimenting and following your suggestions. If we could manage to let it work well then zoneminder on the whole will be better as, IMHO, Raspberry Pi2 is a GREAT platform for managing a small NVR.
If you're not interested in that then simply ignore this message.

knnniggett, 2016-07-07:
The wiki guide you are referring to is written by a community member. If you have questions about that particular guide, the best thing to do is contact the author directly or ask in the user forum.

The Raspberry Pi 2 is one of the platforms I build and test from weekly. ZoneMinder runs well on it as long you understand the limitations of the hardware you are using. The problem you have described is common for a Raspberry Pi. You can read about others with the issues in the user forum, and see the steps they took to get it working.

I've previously recommended to you what you can do to get additional support. It is up to you to decide to follow those instructions.  I will share something with you that I learned myself years ago. If you continue with the mindset that the problem couldn't be caused by you, then you are greatly reducing the chances of solving the problem.