ZoneMinder/zoneminder #1552: Unable to authenticate User

leeyuentuen, 2016-07-07:
I'm getting unable to **authenticate user** in the log.
from server1 I can only see 2 of  6 camera
on server2  I can only see 4 of  6 camera
 when i'm disable the option "system > OPT_USE_AUTH"
the problem is gone
i can see on both server 6 of 6 camera

using zoneminder 1.29.0.
using multiple servers (multi-server)
using ubuntu 16.04 and ubuntu 14.04

```
2016-07-08 01:11:04.318374	zms	10.1.0.3	28345	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:04.307109	undef	10.1.0.3	28345	INF	Server is 2: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:04.273284	zms	10.1.0.3	28343	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:04.265148	undef	10.1.0.3	28343	INF	Server is 2: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:04.239376	zms	10.1.0.3	28341	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:04.233929	undef	10.1.0.3	28341	INF	Server is 2: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:04.212436	zms	10.1.0.3	28338	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:04.201802	undef	10.1.0.3	28338	INF	Server is 2: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:04.201636	zms	10.1.0.3	28337	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:04.195732	undef	10.1.0.3	28337	INF	Server is 2: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:04.138744	zms	10.1.0.3	28335	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:04.134845	undef	10.1.0.3	28335	INF	Server is 2: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:04.120784	zms	10.1.0.3	28333	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:04.116620	undef	10.1.0.3	28333	INF	Server is 2: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:04.087254	zms	10.1.0.3	28331	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:04.079996	undef	10.1.0.3	28331	INF	Server is 2: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:04.061394	zms	10.1.0.3	28329	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.998917	undef	10.1.0.3	28329	INF	Server is 2: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.964960	zms	10.1.0.3	28327	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.951238	undef	10.1.0.3	28327	INF	Server is 2: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.906296	zms	10.1.0.3	28314	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.858543	zms	10.1.0.3	28315	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.833959	undef	10.1.0.3	28314	INF	Server is 2: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.830204	undef	10.1.0.3	28315	INF	Server is 2: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.654491	zms	10.1.0.9	7717	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.650824	undef	10.1.0.9	7717	INF	Server is 3: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.622419	zms	10.1.0.9	7716	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.617894	undef	10.1.0.9	7716	INF	Server is 3: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.597206	zms	10.1.0.9	7715	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.568279	undef	10.1.0.9	7715	INF	Server is 3: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.558466	zms	10.1.0.9	7714	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.538256	undef	10.1.0.9	7714	INF	Server is 3: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.482046	zms	10.1.0.9	7713	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.479299	zms	10.1.0.9	7712	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.474691	undef	10.1.0.9	7713	INF	Server is 3: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.470815	undef	10.1.0.9	7712	INF	Server is 3: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.424311	zms	10.1.0.9	7711	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.414842	undef	10.1.0.9	7711	INF	Server is 3: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.413484	zms	10.1.0.9	7710	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.398665	undef	10.1.0.9	7710	INF	Server is 3: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.327552	zms	10.1.0.9	7709	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.327542	zms	10.1.0.9	7708	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.300765	undef	10.1.0.9	7708	INF	Server is 3: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.299504	undef	10.1.0.9	7709	INF	Server is 3: using Multi-Server Mode.	zm_config.cpp	141
2016-07-08 01:11:03.149109	zms	10.1.0.9	7706	ERR	Unable to authenticate user	zms.cpp	225
2016-07-08 01:11:03.149013	zms	10.1.0.9	7707	ERR	Unable to authenticate user	zms.cpp	225
```

leeyuentuen, 2016-07-07:
When  i'm putting "AUTH_RELAY" to none, the problem will also solved it.

SteveGilvarry, 2016-07-27:
@connortechnology can you reproduce this issue in multi-server environment? Or any clues as to what might be causing it?

SteveGilvarry, 2016-07-27:
#1543 was on experimental branch, but @connortechnology confirmed there that opt_auth working fine for him. Anyone else on multi-server care to comment?

connortechnology, 2016-10-03:
@leeyuentuen Would you be able to test 1.30?

You can use my ZoneMinder ppa at ppa:iconnor/zoneminder

MaLaCoiD, 2016-10-14:
I'm a new ZM user. I created a 16.04 server VM with the 1.30 PPA and get the same error. The leeyuentuen fix works for me, too.

mdetweiler, 2016-11-05:
I see the same thing on a fresh install of Ubuntu 16.04 and ZM 1.29
Followed the easy way wiki to install.  Setting "AUTH_RELAY" to none works, but why?

Is this fixed in 1.30?

fouram, 2016-11-07:
Brand new user on Ubuntu 16.04 fresh install using ppa:iconnor/zoneminder, followed install instructions from wiki. Also getting "'zmc -m 2' exited abnormally, exit status 255" quite a bit

scaprile, 2016-12-19:
Hi, regarding the multi-server (or should I open a nes issue ?):
the docs grant access to zm@localhost, where there should be an entry for each zm@<zmserver> or at least a zm@% if you trust your network:
That is (as showing the idea, not necessarily suggesting to do it this way), substitute
```
mysql -uroot -p -e "grant all on zm.* to 'zmuser'@localhost identified by 'zmpass';"
```
for 
```
mysql -uroot -p -e "grant all on zm.* to 'zmuser'@'zmserver1' identified by 'zmpass';"
...
mysql -uroot -p -e "grant all on zm.* to 'zmuser'@'zmserverN' identified by 'zmpass';"
```
or 
```
mysql -uroot -p -e "grant all on zm.* to 'zmuser'@'%' identified by 'zmpass';"
```

tridiumcontrols, 2017-03-13:
Solved it for me by disabling AUTH_RELAY. 

But if you enable OPT_USE_AUTH, and use default u:admin p:admin then streams work fine, but when you change the password for admin, then things dont work and you get a unable to authenticate user, until AUTH_RELAY is set to NONE, then restarted ZM.

dagremote, 2017-04-15:
Much appreciated - Setting AUTH_RELAY to None worked. Had uninstalled everything and reinstalled and had it working. Didn't tie the issue to the user auth. This was a big help.

ricardobigote, 2017-04-30:
**I've figured out what the problem is!**
This sound's like some 'old' problem, but it was driving me crazy!!! :-P

Someone with the **"Unable to authenticate user zms.cpp"** problem, please confirm if your **AUTH_HASH_SECRET** has some **special characters**, other than letters and numbers! If you do, change it and give it a test with OPT_USE_AUTH enabled and AUTH_RELAY with the hashed option enabled. Don't forget to restart zoneminder after changes.

This worked for me!

mdetweiler, 2017-05-01:
Thanks for letting us all know the real issue @ricardobigote !
I had some other characters in my password so that must have been the issue for me.

ckrzen, 2017-05-23:
@ricardobigote I can confirm that the problem _does_ reappear even if I only use a combination of **letters** & **numbers** in the AUTH_HASH_SECRET. The only solution for me is to set AUTH_RELAY = **none**.

jkeith248, 2017-06-09:
I can also confirm.  Had same issue running Ubuntu 16.04 and ZM 1.30.04.

connortechnology, 2017-06-09:
Ok, tomorrow I'd like the get to the bottom of this.  Would help if some of the people afflicted were available, so I can duplicate your setup and figure it out.

kz, 2017-06-14:
I don't use special characters here. However, disabling AUTH_RELAY allows me to view my stream again. Ubuntu 16.04, v1.30.4.

gbengeult, 2017-08-04:
I can confirm that this problem still exists, at least for me. I don't use special characters in AUTH_HASH_SECRET, I don't have a multi-server setup, and I am now running v1.30.4, though the problem also existed for me in v1.30.2.

I have some additional information about the bug. It did not occur until I added a second camera to my configuration. Now, even if I delete the second camera (and restart zm) the problem persists. I hope this observation helps to track it down.

phillijw, 2017-08-16:
The problem is happening for me as well with a fresh install of ubuntu 16 and only 1 camera. Secret only has lowercase, uppercase, and numbers. No special characters. Problem goes away if I uncheck OPT_USE_AUTH or set AUTH_RELAY=None.

version 1.30.4

AndreyTsl, 2017-08-21:
I had the same issue due to different timezones on system (Ubuntu Server 16.04) and Apache. Once I set same timezones problem disappeared.

brunojjean, 2017-08-28:
Same issue with RHEL 7.4, v1.30.4, fresh install, one monitor only, only uppercase, lowercase letter and number in both AUTH_HASH_SECRET & admin password. Server located behind a router with port forwarding and accessed via Internet. Client browser is Chrome on Windows 10. Problem goes away if I set AUTH_RELAY=None.

mdetweiler, 2017-08-28:
As a data point.  I did not see this issue on my recent clean install of Ubuntu 16.04 and ZM 1.30.4.
I followed the install wiki and used a password with both lowercase and uppercase letters, numbers and a hyphen. 

Config variables were set to
OPT_USE_AUTH = checked
AUTH_TYPE = builtin
AUTH_RELAY = hashed 
AUTH_HASH_SECRET  = ...Change me to something unique...
AUTH_HASH_IPS = checked
AUTH_HASH_LOGINS

brunojjean, 2017-08-28:
My bad, I did not logout from the console after changing the password and removing the special character.  It is working now with or without special characters. Looks like there is a timeout on the live stream page while the main console is still working like a charm even staying logged in for more than 24 hours.

ineedfat, 2017-11-10:
I had this problem and I was able to resolve the problem.

Because the auth key is an md5 hash of of AUTH_HASH_SECRET + Username + *PASSWORDHASH + TIME + DATE. Depending on your server and client, the TIME and Date may be off. For example, my server was set to PST time, but the client was using Mountain time for its hash calculation. During half of the year, it would work fine until daylight savings hit and I was off by 1 more hour. It was only working before because Zoneminder hash checking allows up to one hour difference. Daylight savings increased this to 2 hours, and I ended up getting "unable to authenticate user". Setting the server time clock to Mountain time solved my issue.

You can enable debug logging to see the output of hash calculations.

phillijw, 2017-11-10:
If that's really the issue, isn't UTC time an easy fix?

connortechnology, 2017-11-10:
It's less important what the time zone is, and more than everyone is in the same time zone. When I say everyone I mean the server running the php code and the database. Because we use the SQL NOW function, the database has to be configured to be in the same time zone as the php code.

undigo, 2018-03-12:
@ineedfat Right solution,
change Timezone of server to my region with this guide:
[https://www.thegeekstuff.com/2010/09/change-timezone-in-linux/](https://www.thegeekstuff.com/2010/09/change-timezone-in-linux/)
ZM config:
OPT_USE_AUTH = checked
AUTH_TYPE = builtin
AUTH_RELAY = hashed
AUTH_HASH_SECRET = notspecialcharacters
AUTH_HASH_IPS = checked
AUTH_HASH_LOGINS = checked
Hardware / OS 
Linux cubietruck 4.14.18-sunxi #24 SMP Fri Feb 9 16:24:32 CET 2018 armv7l GNU/Linux
PHP 7.0.27-0+deb9u1 (cli) (built: Jan  5 2018 13:51:52) ( NTS )
Copyright (c) 1997-2017 The PHP Group
Zend Engine v3.0.0, Copyright (c) 1998-2017 Zend Technologies
    with Zend OPcache v7.0.27-0+deb9u1, Copyright (c) 1999-2017, by Zend Technologies
Server version: Apache/2.4.25 (Debian)
Server built:   2017-09-19T18:58:57
ZM version 1.30.4
not restart ZM server necessary.
the most harder is compatibility php 7 with apache 2, not ZM :)

DerVerruckteFuchs, 2018-04-07:
I'm running v1.30.4 on Debian 9.4 with 4 cameras. I've had a similar issue. None of my cameras feeds are visible and give those image not found/link is broken icons with AUTH_RELAY set to plain or hashed. Setting AUTH_RELAY to none fixes the issue. However, users without special characters in their password can view the camera feeds with AUTH_RELAY set to hashed. I replaced my own password with one without special characters and it worked fine with AUTH_RELAY set to hashed and AUTH_HASH_SECRET can have special characters without issue. @brunojjean seems to have found out the same thing, but wasn't as clear with their explanation.

ZM config:
OPT_USE_AUTH = checked
AUTH_TYPE = builtin
AUTH_RELAY = hashed
AUTH_HASH_SECRET = HasSpecialCharacters
AUTH_HASH_IPS = checked
AUTH_HASH_LOGINS = checked

Important to note I'm still getting the `zms Unable to authenticate user` errors in my log after the changing my password and having everything working as expected. I'm also getting a `web-php Socket /var/run/zm/zms-XXXXXXs.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.` error. I've noticed that `/var/run/zm/zms-XXXXXXw.sock` exists, but not `zms-XXXXXXs.sock`. I'm guessing the socket naming convention changed somewhere, or it's related to the zms authentication issue.

DerVerruckteFuchs, 2018-04-07:
Does the live stream backend stuff rely on zmu at all? There's an awfully similar issue with #464 and special characters. If #464 gets fixed it may fix this issue as well.

closed 2018-06-22 by stale[bot]

giuarb, 2018-11-14:
Ubuntu 18.04 Server, 1.32.2 Same problem after changing the Time in php 
set on UTH_RELAY=None. for now. and all camera works. However not sure if it is related but I can't play any recordings

kvacha13, 2018-12-13:
You can try:
1. configure in ZM web console: Options->System:
OPT_USE_AUTH = checked
AUTH_TYPE = builtin
AUTH_RELAY = hashed
AUTH_HASH_SECRET = “I had set it to my zm admin password from mysql users but will work any word”
AUTH_HASH_IPS = checked
AUTH_HASH_LOGINS = checked

2. Save + Restart zoneminder
3. AUTH_HASH_SECRET  = “”     (this time set it to blank)
6. Save+restart zoneminder.

kvacha13, 2018-12-14:
No stream when OPT_USE_AUTH enabled and AUTH_RELAY set to hashed:
1.	First just recheck compliance with the following installation instruction :(https://bkjaya.wordpress.com/2018/05/22/how-to-install-zoneminder-v1-31-44-on-ubuntu-18-04-lts/ ). Especially check the mysql zm database users (https://askubuntu.com/questions/110841/how-to-reset-zoneminder-password):

$ mysql -u root -p
Enter password:
$ show databases;
$ use zm;
$ select * from Users;

Check if you are using the hashed password.
If everything is ok here then, follow below to fix the problem:

2.	The following can cause the server hash problem (thanks to https://forums.zoneminder.com/viewtopic.php?t=23148):

- The server clock is more than 1 hour off
- The timezone in php.ini is not set at all or incorrectly
- AUTH_HASH_IPS is on, but the server cannot determine the IP of the remote client

And the most importan part here seems to be timezone as ZM_AUTH_RELAY explanation says (see ZM>options> ZM_AUTH_RELAY> ?):
ZM_AUTH_RELAY
When ZoneMinder is running in authenticated mode it can pass user details between the web pages and the back end processes. There are two methods for doing this. This first is to use a time limited hashed string which contains no direct username or password details, the second method is to pass the username and passwords around in plaintext. This method is not recommend except where you do not have the md5 libraries available on your system or you have a completely isolated system with no external access. You can also switch off authentication relaying if your system is isolated in other ways.

3.	To check the server clock, set Camera to modect and record event (it would be nice having server timestamp in zm console). Then check the recorded time against your system local time. If different, set the correct server timezone:
Using the Command Line (terminal)
Using the command line, you can use sudo dpkg-reconfigure tzdata.
-	Open a terminal window by going to Applications>Accessories>Terminal
-	sudo dpkg-reconfigure tzdata
-	Follow the directions in the terminal.
-	The timezone info is saved in /etc/timezone - which can be edited or used below

4.	After setting correct timezone recheck it (see 3) and if ok restart all service from terminal:
  -sudo service zoneminder stop
  -sudo service apache2 stop
  -sudo service mysql stop
   -and then run all services: sudo service mysql start  …..apache2 start….zonemidner start

Hope it works. Worked for me.

didi1357, 2019-03-10:
I also had the same issues as described here. I verified/fixed the following:
+Correct timezone in php.ini
+sudo dpkg-reconfigure tzdata
+OPT_USE_AUTH = checked
+AUTH_TYPE = builtin
+AUTH_RELAY = hashed
+AUTH_HASH_SECRET = No special characters
+AUTH_HASH_IPS = unchecked
+AUTH_HASH_LOGINS = unchecked
Up to this time I had still no success. What finally fixed it for me was reducing the length of:
+AUTH_HASH_SECRET to 6 characters
+admin password to 6 characters

Greetings

realknotworking, 2019-03-12:
My zoneminder 1.32.3 install on Cent7 was working fine, my issue was with zmNinja Pro. **I was able to use the app without issue while I was on my local network**, but when I left home and switched to my phone's internet, the authentication user errors kept me from logging in. My server timezone was correct; following didi's other tips above I checked/unchecked the specific inputs and limited the password & hash secret to 6 characters in ZM (remember to change the password in the app, too). When I opened the app again, it complained that I needed AUTH_HASH_LOGINS enabled, I checked that config box in ZM, saved/restarted ZM, and now the zmNinja app is working on both my phone & local wifi.

Weird that the app login worked on the local wifi network but not on the phone's connection......

didi1357, 2019-03-13:
@realknotworking 
Have you had AUTH_HASH_IPS enabled? I guess they hash the client IP somehow if this is checked. If your ZM instance is behind a NAT, it may believe the access is originated from your router as the client and use this for hash calculation, where the app uses your public IP from the internet, or maybe also a NATed one since carrier grade NAT got quite popular for mobile connections.

Ilia-SB, 2019-05-15:
Had this problem, tried everything and found a solution by pure luck.
My php.ini had another entry of date.timezone at the very end of the file and naturally it was set to Etc/UTC. Changed that to my timezone and voila!
Had to enter new AUTH_HASH_SECRET and password in ZM options. Guess the hash had to be regenerated. Rebooted the server to be sure.

I guess a good way to make sure yor php is properly configured is to look at the output of phpinfo(). To do so
- in browser go to http://your_server_ip. The Apache default page should appear. Note www root directory listed there (/var/www/html/ in my case)
- create a file in that directory (for example /var/www/html/phpinfo.php) and put <?php phpinfo(); ?> there
- now go to http://your_server_ip/phpinfo.php and check time zone in the output