ZoneMinder/zoneminder #1554: Unable to connect to database

0xABAB, 2016-07-08:
`zmaudit` adds `Unable to connect to database` to my log every 15 minutes. I can connect to the database just fine (as root) and didn't change anything about this setup. It has worked before. After `service zoneminder restart` it does work. 

Operating system Ubuntu 14.04.4 LTS. 
zoneminder                1.29.0-trusty     amd64

Additional symptoms are the absence of the name of the zone causing an alarm; I'd expect log additional log lines at the very least for cases when an empty string is used as the name of a zone. It would be better if it would be impossible by design for that to happen, however. 

I'd be interested in running 1.30, but it is not released for my operating system.

knnniggett, 2016-07-08:
Sorry you are having an issue, but you have not presented any data to indicate this is a bug with the source code. The ZoneMinder 1.29.0 package you are using has been well tested with no database connection issues. To get support for this, create a thread in the user forum. In addition to what you've stated, we will also need to know exactly which instructions you followed to install ZoneMinder on your system. You can expect to be asked.

closed 2016-07-08 by knnniggett

0xABAB, 2016-07-08:
Obviously I followed the instructions to the letter of whoever provided that package (AFAIK, there is only one person on this planet providing packages now in this configuration in a public fashion, which makes your complaint rather retarded). 

ZoneMinder has historically been a buggy piece of shit, so to even say that it has been "well-tested" is a joke. 

Let's say I do find the bug and patch it, are you going to change the homepage to say that "ZoneMinder is a piece of shit which nobody in their right mind should ever use"? No? Why should I spend time fixing your amateuristic hobby project then?

I'd much rather just try Motion and see whether that also crashes within 100 hours of operation. I think it's an absolute disgrace to write empty strings to a database when there are no zones defined which are empty string. That wouldn't happen if you would know what you are doing. 

My current solution is to just automatically restart it when it outputs the error message, which means that I will lose data, which means that ZoneMinder can't actually mind zones; it can mind zones for less than 100 hours. 

On the ZoneMinder website you are claiming that ZoneMinder is `A full-featured, open source, state-of-the-art video surveillance software system`. That's completely wrong. 

The state of the art in video surveillance can actually detect humans, figure out what is a shadow, detect cats, do threat detection, make use of GPUs, etc. ZoneMinder is just a trivial motion detection algorithm with a CRUD interface. What were you smoking when you put that on the website?

knnniggett, 2016-07-08:
@0xABAB Your immature temper tantrums will not help you solve your problem and will be deleted, wherever you decide to post them. Thread locked.