ZoneMinder/zoneminder #1555: SQL Errors Adding Monitor

mochoandre, 2016-07-09:
I have done a source instalation using this guide adapting for ubuntu 16.04 (with php7.0)

https://github.com/pliablepixels/zmNinja/wiki/Hard-Way:-Ubuntu-15,-Compile-ZoneMinder-API-from-source

However when i try to add a monitor it fails with MySQL Erros

/usr/local/share/zoneminder/www/includes/database.php     line 115

`SQL-ERR 'SQLSTATE[HY000]: General error: 1366 Incorrect integer value: '' for column 'V4LMultiBuffer' at row 1', statement was 'insert into Monitors set LinkedMonitors = '', Name = 'Monitor-1', ServerId = '1', Type = 'Remote', Function = 'Monitor', Enabled = '1', RefBlendPerc = '6', AlarmRefBlendPerc = '6', AnalysisFPS = '20', MaxFPS = '25', AlarmMaxFPS = '25', Device = '/dev/video0', Channel = '0', Format = '255', Palette = '0', V4LMultiBuffer = '', V4LCapturesPerFrame = '1', Options = '', LabelFormat = '%N - %d/%m/%y %H:%M:%S', LabelX = '0', LabelY = '0', LabelSize = '1', ImageBufferCount = '50', WarmupCount = '25', PreEventCount = '25', PostEventCount = '25', StreamReplayBuffer = '1000', AlarmFrameCount = '1', EventPrefix = 'Event-', SectionLength = '600', FrameSkip = '0', MotionFrameSkip = '0', AnalysisUpdateDelay = '0', FPSReportInterval = '1000', DefaultView = 'Events', DefaultRate = '100', DefaultScale = '100', WebColour = 'red', Exif = '0', SignalCheckColour = '#0000c0', Protocol = 'http', Method = 'simple', Host = '192.168.1.44', Port = '8080', Path = '/video', Colours = '3', Width = '1280', Height = '720', Orientation = '0', Deinterlacing = '0', Sequence = 1'`

/usr/local/share/zoneminder/www/includes/database.php     line 115

`SQL-ERR 'SQLSTATE[HY000]: General error: 1366 Incorrect integer value: '' for column 'ServerId' at row 1', statement was 'insert into Monitors set LinkedMonitors = '', Name = 'HTCOne', ServerId = '', Type = 'Remote', Function = 'Monitor', Enabled = '1', RefBlendPerc = '6', AlarmRefBlendPerc = '6', AnalysisFPS = '', MaxFPS = '', AlarmMaxFPS = '', Device = '/dev/video0', Channel = '0', Format = '255', Palette = '0', V4LMultiBuffer = '', V4LCapturesPerFrame = '1', Protocol = 'http', Host = '192.168.1.44', Port = '8080', Method = 'simple', Options = '', Path = '/video', User = '', Pass = '', Colours = '3', Width = '320', Height = '240', Orientation = '0', Deinterlacing = '0', RTSPDescribe = '', LabelFormat = '%N - %d/%m/%y %H:%M:%S', LabelX = '0', LabelY = '0', LabelSize = '1', ImageBufferCount = '50', WarmupCount = '25', PreEventCount = '25', PostEventCount = '25', StreamReplayBuffer = '1000', AlarmFrameCount = '1', SignalCheckColour = '#0000c0', EventPrefix = 'Event-', SectionLength = '600', FrameSkip = '0', MotionFrameSkip = '0', AnalysisUpdateDelay = '0', FPSReportInterval = '1000', DefaultView = 'Events', DefaultRate = '100', DefaultScale = '100', WebColour = 'red', Sequence = 1'`

regards

SteveGilvarry, 2016-07-09:
General support is provided through forums.zoneminder.com.
http://zoneminder.readthedocs.io/en/latest/installationguide/ubuntu.html#easy-way-ubuntu-16-04
MySQL 5.7 issue being worked around. Details in link above.

Also even if you want to build from source build as a package as perl does not make uninstall correctly.

closed 2016-07-09 by SteveGilvarry

mochoandre, 2016-07-09:
Ok, that did solve the issue.
Thanks

rnveach, 2017-11-05:
I followed the easy way documentation for 16.04 for a fresh ZM install including the `NO_ENGINE_SUBSTITUTION` and I still get the error mentioned above.
I made the mysql changes after ZM was installed but have restarted mysql and rebooted with no difference.
Is there something else I could be missing?

NicolasGoeddel, 2018-05-23:
I don't understand why this issue was closed. I installed zoneminder 1.30.4 on Ubuntu 16.04 from repository and the bug is still there. The script still tries to write an empty string into an nullable integer field.